don/cordova-plugin-ble-central #660: Promise from writing to a characteristic does not return a response on success

odimou, 2019-01-25:
I'm writing a Uint8Array to a characteristic on my device with success. Writing to the characteristic should return a response. However what I'm getting as a result is "OK" for Android and null for iOS. Here's my code:

   ```
   this.ble.write(deviceID, serviceUUID, characteristicUUID, uint8ArrayData.buffer)
          .then(resultWrite => {
            console.log(resultWrite);
          });
    ```

The result is "OK" on Android and null on iOS.
Is there a change that should be made (e.g. in the Peripheral.java for Android), in order to make that work?

Thanks

closed 2019-01-31 by odimou

StephanBis, 2020-08-17:
I'm experiencing the same issue, is there something you changed to make this work?

JEricaM, 2021-05-03:
Same problem here!