don/cordova-plugin-ble-central #663: Can not connect with my device (headset)

marioshtika, 2019-02-20:
Hello I am trying to connect with my bluetooth device, which in the available devices in Android settings it looks like a headset (even it is not a headset).
[image]

Is there a way to connect with this type of bluetooth device?

droghio, 2019-03-07:
Are you trying to a classic Bluetooth device or BLE, can you provide specs?

marioshtika, 2019-03-07:
It's a glucometer and I think it is BLE (I am not sure though).
I think it is using a different [profile](https://developer.android.com/guide/topics/connectivity/bluetooth#Profiles).
It looks like it is using the "Headset" profile, instead of "Health Device" profile.

Could the usage of a "Headset" profile, be the reason why I can not connect using this plugin?

Thank you in advance

marioshtika, 2019-03-07:
I just find out that the device I am using it has Bluetooth v3 and only supports serial communication but not HDP.
Can I use this plugin to communicate with my device?

droghio, 2019-03-08:
Unfortunately no. BLE is only available on certain BT4+ devices. Bluetooth 3 is exclusively classic and not compatible with this library. Despite the naming similarity the two are completely different protocols. There should be a few serial terminal apps in the android store that will let you interface with the device.

don, 2019-10-17:
On **Android** you can use [cordova-plugin-bluetooth-serial](https://www.npmjs.com/package/cordova-plugin-bluetooth-serial) if the device supports RFCOMM

closed 2019-10-17 by don