don/cordova-plugin-ble-central #664: Write Without Response, more than 20 bytes not working, after MTU is 512.

chaymankala, 2019-02-22:
As soon as the device is connected to a peripheral, I am trying to set MTU Size to 512(as my message is long, its around 250 bytes). The callback says "OK" (success). But, still the message is being chopped when I see in console of my BLE peripheral.

Here's my code.
```
    setMTUSize(mac_id) {
        return new Promise((_, d) => {
            this.bs.requestMtu(mac_id, 512)
                .then(success => {
                    console.log("MTU SET and waiting");
                    setTimeout(() => {
                        console.log("MTU SET and done waiting");
                        return _(success);
                    }, 500);
                }).catch(e => d(e));
        })
    }
```

**PS**: 1. I used delay, just for testing. I tried without delay also. It was not working in both of the cases.
       2. I also tried setting the MTU to 250, 200. But the message is still sent in 20 bytes. 

Please help me. Thanks in Advance.

woodz-, 2019-02-27:
Which version of cordova and this ble central plugin are you using?
Are you able to test on Android and IOS, then tell about your experience?

chaymankala, 2019-02-27:
@woodz-
 I'm using the latest ble-central plugin (1.2.2),
 Cordova: 8.1.2

I tested on my OnePlus 5 (Android 9). Every time on writeWIthoutResponse, it sends 20 bytes only. I could confirm this from the console of my BLE peripheral (which is true for sure). I even tried sending the same payload using Bluetooth Serial Monitor (Android App) on my phone, which works perfectly fine. 
Coming to iOS, I can't test as of now, because the simulator wont support BLE usage from MacBook, anyway, Im only concerned about android right now.

Any help would be appreciated. TIA :)

woodz-, 2019-02-27:
Are you able to test on a lower android version (older phone or custom rom), 7.1 would be a good candidate?
Although you don't target to IOS, it would be good to see, whether the problem is rather on the peripheral or on the plugin software.
Such tests help to localize problems.

chaymankala, 2019-02-27:
Sure, I will test and let you know soon @woodz-

chaymankala, 2019-03-01:
@woodz- 
Seems like HM-10 doesnt support MTU more than 20 Bytes.
Figured out some other way to send 20 Bytes in chunks in a queue. It's working fine.
So, closing this issue.

closed 2019-03-01 by chaymankala