don/cordova-plugin-ble-central #667: Connection callback does not fire after Android update

FLATT-Dev, 2019-03-04:
This is on Samsung Galaxy S3,  The plugin works fine on other phones I have
My application was working just fine until 2 days ago when I got a prompt to update the system from my  provider (AT&T) So I did. That did several Android updates as well
Now the plugin connect callback never fires.
I added additional logging to Peripheral.connect and Peripheral.onConnectionStateChange
Here is what I get the log from Peripheral.connect method
03-04 15:02:49.361  7915  7915 D Peripheral: Connecting to 00:A0:50:28:37:29
03-04 15:02:49.361  7915  7915 D Peripheral:  using API > 23, transport AUTO
03-04 15:02:49.416  7915  7915 D Peripheral: after connectGatt, got GATT object {"name":"ST1501","id":"00:A0:50:28:37:29","advertising":{"CDVType":"ArrayBuffer","data":"AgEGAgoAGP8AAAAAAAAAAAAAAAA3KQAgt6cc92QDAAcJU1QxNTAxFv8AAP3NAAAAAAAAAAAAAAAAAAAAAAA="},"rssi":-39,"services":[],"characteristics":[]}
But none of the log messages I added to Peripheral.onConnectionStateChange are printed so it appears that the callback never fires.
I control the beacon firmware and I do see the GATT connection indication event fires in the firmware and its BLE stack is in the connected state.

Any ideas??
Thanks
Andy

Here is what my "About Device" page shows now (not sure what it was before)
Android version : 6.0.1
Android security patch level
July 1,2017
Kernel version
3.10.9-7663112
dpi@SWDG4714 #1 
Fri Jun 30 13:21:03 KST 2017
Build Number
MMB29K.J320AUCS3AQG1

droghio, 2019-03-24:
When you build for Android do you know what version of Android Studio cordova is using in the background? It might be up need to update cordova or Android Studio to remain compatible with the newer Android version.

FLATT-Dev, 2019-03-26:
Updated to Cordova v 9, Android Studio 3.3.2
Made no difference
Andy

don, 2019-10-18:
@FLATT-Dev I think this is a device specific issue which is really difficult to fix without the hardware. Hopefully you have a different phone by now.

closed 2019-10-18 by don