don/cordova-plugin-ble-central #669: Data lost when sending large data

ca-kou, 2019-03-14:
I'm using this code to send large data https://gist.github.com/don/ad3d6f444925c57d169baf54e262348f

The MTU is 185, so I set the MAX_DATA_SEND_SIZE to 180.

I'm sending data from iPad to a esp32 board using gatt.
If I send 180*70 bytes data, I can only get 180*60 data in esp32 side.

How can I solve the problem?

droghio, 2019-03-24:
That is a huge MTU, did you see any warnings or errors on the esp32 when you attempted the transfer? You might have to break up the data read into multiple transactions.

Also what do you mean by 180*70 bytes? Are you saying you are trying 70 transactions each with 180bytes?

don, 2019-10-17:
You shouldn't need to mess with the MTU. iOS and Android should chunk data and send transparently using blobs. Each platform has limits for the max characteristic size. In the past iOS was 500 and Android was about 100 bytes. It might be higher now, but many peripherals have lower max amounts. Depending on the hardware on both sides, there will be some max limit. Over that limit your app needs to be responsible for breaking data up into multiple writes to a characteristic. BLE doesn't really work well for streaming large amounts of data across a characteristic.

closed 2019-10-17 by don