don/cordova-plugin-ble-central #670: Android 64 bits before August 1st ?

favnec5, 2019-03-22:
Hello,
Google has [announced](https://android-developers.googleblog.com/2019/01/get-your-apps-ready-for-64-bit.html) that beginning August 1st, 2019, all new apps and app updates must provide a 64-bit version in addition to a 32-bit version. 
What, specifically, does this plugin need to do to meet this requirement ?
Is there an update planned for compatibility?
Best reguards

droghio, 2019-03-24:
Pretty sure that announcement only applies for native code, i.e. C/C++. This library is written in Java, so I don't believe it applies. I could be wrong though.

favnec5, 2019-03-27:
thanks a lot

don, 2019-10-17:
@favnec5, @droghio is right, that doesn't apply to this plugin since it's java

closed 2019-10-17 by don