don/cordova-plugin-ble-central #671: stopNotification() on ios can cause an exception

domonoky, 2019-03-25:
It looks like calling stopNotification on ios can cause the following exception:

*** Assertion failure in -[CBUUID initWithString:], /BuildRoot/Library/Caches/com.apple.xbs/Sources/MobileBluetoothFramework/MobileBluetooth-123.1/CoreBluetooth/CoreBluetooth/CBUUID.m:158
*** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'String (null) does not represent a valid UUID'

I thinks this may be caused that a new notifcation arrives after the stopNotification call.

If i remove the call to stopNotification() and just disconnect, everything works fine.

Ariaal, 2019-07-04:
hi ,
When I use startNotification, it will report an error.  like this "Looking for 569A2000-B87F-490C-92CB-11BA5EA5167C with properties 16"   
ios 12.3.1 
Do you have any suitable solutions or how do you do it?
thanks!

don, 2019-10-18:
@domonoky I can't duplicate this. FWIW I never stop notifications, I just disconnect. I'm closing this issue since it's old. If you experience this again please open a new issue with more details on how to duplicate.

@Ariaal it sounds like you're trying to startNotification on a characteristic that doesn't support notification.

closed 2019-10-18 by don