don/cordova-plugin-ble-central #679: BLE scanning with Ionic4: not scanning

aircable, 2019-05-03:
Scanning does not work with Ionic4 (4.12.0) simple test app. There is no error message, enable checks are coming back positively. The subscription just never returns. See logcat below. Any help is really appreciated. 

Here is how to reproduce the problem: 

ionic start BLEscanner (chose blank and don't connect)
cd BLEscanner/
cordova plugin add cordova-plugin-ble-central --variable BLUETOOTH_USAGE_DESCRIPTION="Scan for mesh packets"
npm install --save @ionic-native/ble
ionic cordova platform add android@8.0.0 (tried with 7.1.0 and 6.4.0)
ionic cordova plugin add cordova-plugin-ionic-webview@latest

This is the code for home.page.ts:

import { OnInit, Component } from '@angular/core';
import { BLE } from '@ionic-native/ble/ngx';
import {
  Platform,
} from '@ionic/angular';

@Component({
  selector: 'app-home',
  templateUrl: 'home.page.html',
  styleUrls: ['home.page.scss'],
})
export class HomePage implements OnInit  {

  constructor(
    private ble: BLE,
    private platform: Platform,
  ){
    console.log("constructor home page");
  }
  async ngOnInit() {
    // Since ngOnInit() is executed before `deviceready` event,
    // you have to wait the event.
    await this.platform.ready();
    console.log(`enter view, start scanning`)
    this.ble.scan([], 5).subscribe(
      device =>
        console.log(JSON.stringify(device))),
      error => console.log(`error ${error}`);
  }
}

compile and run:

ionic cordova run android --device
check chrome://inspect/#devices for inspect and look at the console.
All you get is:
Angular is running in the development mode. Call enableProdMode() to enable the production mode.
cordova.js:1218 deviceready has not fired after 5 seconds.
vendor.js:74395 Ionic Native: deviceready event fired after 3303 ms
home-home-module.js:98 constructor home page
home-home-module.js:135 enter view, start scanning

Yes, there are many BLE advertisers present.

The error message in android logcat is:
05-03 15:30:42.368 17937 20508 I Places  : ?: PlacesBleScanner stop()
05-03 15:30:42.394 17937 17937 I BeaconBle: 'L' hardware scan: scan stopped, no clients
05-03 15:30:42.394 17937 17937 I BeaconBle: Scan canceled successfully.
****

halindrome, 2019-05-04:
I am using this plugin with ionic with no problems. Is it possible you did not request permission to access location services?   I think that is what is required to get at the bluetooth devices.

aircable, 2019-05-04:
Don't get me wrong, I have used this plugin before with great success.
Just lately with the new version, it stopped working.

Just try these few commands on your machine with Ionic 4.12.0.

And yes, the location permissions are set. It even reports that the plugin is 
enabled and active:

private checkBluetooth() {
    this.ble.isEnabled().then(
      success => {
        console.log("Bluetooth is enabled");
      },
      error => {
        this.loading.dismiss();
        this.showToast(`Bluetooth is *not* enabled`);
        this.enableBluetooth();
      }
    );
  }

  private enableBluetooth() {
    this.ble.enable().then(
      success => {
        console.log("Bluetooth is enabled");
      },
      error => {
        this.showToast("The user did *not* enable Bluetooth");
      }
    );
  }

aircable, 2019-05-07:
The answer is: version mismatch. If @ionic-native/core and @ionic-native/ble are not the same it stopped working. Now with 5.5.1 for both it is fine.

closed 2019-05-07 by aircable