don/cordova-plugin-ble-central #682: Non-GATT Device

markarupert, 2019-05-10:
I have a device that is non-GATT will this plugin still work with that?

hervehobbes, 2019-05-17:
no

aircable, 2019-05-17:
or yes? We do Bluetooth mesh communication with our non-GATT devices.
The idea is to use the ble-central plugin for receiving mesh packets and use 
the ibeacon-plugin (for now) to transmit specially formed iBeacons containing
a mesh packet to the device. That works on Android and iOS.

droghio, 2019-05-29:
This plugin is most relevant for communicating with GATT servers. As @aircable mentioned it can also read advertisement/scan response data but in my opinion you would be better off leveraging other plugins if that is what you are going for. This plugin does not work with bluetooth classic devices so if you are trying to connect to a bluetooth stereo for example you would need to look elsewhere.

don, 2019-10-17:
@aircable That's awesome to hear you've made this work with Mesh
@markarupert if you can make it work for your needs great, typically I'd recommend writing a new plugin for your specific needs. (I write non public cordova plugins to very specific problems for clients.)

closed 2019-10-17 by don