don/cordova-plugin-ble-central #686: Read large data with startNotification

neumotngayem, 2019-05-21:
Hi I am using startNotification to get the data from characteristic, but It just return the ArrayBuffer with the length is 20, while my data is around 47 byte.

Ariaal, 2019-05-22:
you can get data three times.  Low-power Bluetooth does not exceed at most each 20byte . by the way,how can you get data ? is using startNotification ? why i can not get my data ?where you uesd it ?

Ariaal, 2019-05-22:
you can get data three times. Low-power Bluetooth does not exceed at most each 20byte . by the way,how can you get data ? Did you use startNotification ? why i can not get my data ?where you uesd it ? i uesd it when i sent my data ,but it does not matter.

don, 2019-10-18:
Not enough info here to debug. It sounds like the characteristic length might be 20 bytes.

closed 2019-10-18 by don