don/cordova-plugin-ble-central #687: StartNotification Problem

FabioPuviani, 2019-05-22:
Hi,

Thank's for this great plugin.

On IOS 12, if i use startNotification, Xcode logged this error:
"Client application failed to make a policy decision via WebPolicyDecisionListener"

If i try tu put random parameters inside the function, startNotification use the error callback, as i aspected.

Ariaal, 2019-07-04:
I also encountered this problem.
Tell me if you solve it!
Thanks!

FabioPuviani, 2019-07-04:
No not yet, I think I'll switch to another plugin

Ariaal, 2019-07-05:
Do you add  cordova-custom-config ?

FabioPuviani, 2019-07-08:
No, but i don't need background notification

Ariaal, 2019-07-08:
The displayed write was successful, but my device failed to accept and no data was passed.

LuchoManto, 2019-07-10:
i added custom-config, but startNotification still fails. I don't get that error though. Every time i attempt to run startNotification, the failure function executes.

don, 2019-10-17:
The characteristic needs to support notifications or indications in order for startNotification to work. The peripheral controls if the characteristic is read, write, notify, etc.

closed 2019-10-17 by don