don/cordova-plugin-ble-central #691: ble.startScan does not work on iOS 12.3 or greater

sravetta, 2019-06-19:
Hi, I'm using this code to scan all BLE devices in my Cordova app developed for iOS and Android:
```javascript
var deviceName = "<name of device>";
var devices = [];

ble.startScan(
    [],

    function (dev) {
        console.log("Discovered device", dev);
        if (dev.name === deviceName) {
            console.log("--- My device ---");
            this.devices.push(dev);
        }
    }.bind.this(),

    function (err) {
        console.log("Error", err);
    }.bind(this)
)
```
when I run the code snippet mentioned above and a device is discovered, the app suddenly crashes without neither a success nor an error message.

This happens only with devices running iOS 12.3 or greater. Also notice that if no device is detected when running the code, the app doesn't crash. This issue is only happening while running the app under iOS 12.3 or greater (iPhone 8 and iPhone X): on previous version of iOS (eg. 12.1) or on any version of Android it's working fine.

cairinmichie, 2019-06-20:
Are you sure your permissions are setup correctly in the info.plist? This looks like a permissions issue.

Also shouldn't `}.bind.this(),` be `}.bind(this),`? In the success callback.

sravetta, 2019-06-20:
I wrote it correctly in the effective code, however while rewriting it when opening the issue I had it wrong.

By the way, even after multiple tests, I am pretty confident to say that the problem is very similar to this https://github.com/don/cordova-plugin-ble-central/issues/685. The fix suggested  for such issue is to remove a variable `kCBAdvDataLeBluetoothDeviceAddress` but it is a hack. Sadly, I have no experience with objective C to solve it. Have you got suggestions about how that issue can be solved?

That said, I will also follow the way you suggested about changing the permissions, just in case those are involved in some way.

ragcsalo, 2019-08-19:
Any progress/solution regarding this bug?

pacMakaveli, 2019-09-06:
I hope my issue and this are related. Further more, here's a stacktrace from xcode while running on device:

```2019-09-06 12:53:30.774805+0100 MyApp[9466:1237796] Apache Cordova native platform version 4.5.5 is starting.
2019-09-06 12:53:30.774893+0100 MyApp[9466:1237796] Multi-tasking -> Device: YES, App: YES
2019-09-06 12:53:30.780377+0100 MyApp[9466:1237796] CDVWKWebViewEngine: trying to inject XHR polyfill
2019-09-06 12:53:30.821606+0100 MyApp[9466:1237796] CDVWKWebViewEngine will reload WKWebView if required on resume
2019-09-06 12:53:30.821673+0100 MyApp[9466:1237796] Using Ionic WKWebView
2019-09-06 12:53:30.822014+0100 MyApp[9466:1237796] [CDVTimer][console] 0.051975ms
2019-09-06 12:53:30.822109+0100 MyApp[9466:1237796] [CDVTimer][handleopenurl] 0.054002ms
2019-09-06 12:53:30.823535+0100 MyApp[9466:1237796] [CDVTimer][intentandnavigationfilter] 1.402974ms
2019-09-06 12:53:30.823606+0100 MyApp[9466:1237796] [CDVTimer][gesturehandler] 0.042081ms
2019-09-06 12:53:30.825536+0100 MyApp[9466:1237796] [CDVTimer][file] 1.899004ms
2019-09-06 12:53:30.825699+0100 MyApp[9466:1237796] CDVIonicKeyboard: resize mode 1
2019-09-06 12:53:30.826219+0100 MyApp[9466:1237796] [CDVTimer][cdvionickeyboard] 0.641108ms
2019-09-06 12:53:30.829484+0100 MyApp[9466:1237796] [CDVTimer][splashscreen] 3.242016ms
2019-09-06 12:53:30.834509+0100 MyApp[9466:1237796] [CDVTimer][statusbar] 4.958987ms
2019-09-06 12:53:30.834947+0100 MyApp[9466:1237796] Bluetooth Serial Cordova Plugin - BLE version
2019-09-06 12:53:30.834963+0100 MyApp[9466:1237796] (c)2013-2014 Don Coleman
2019-09-06 12:53:30.836326+0100 MyApp[9466:1237796] [CDVTimer][bluetoothserial] 1.771927ms
2019-09-06 12:53:30.836703+0100 MyApp[9466:1237796] Cordova BLE Central Plugin
2019-09-06 12:53:30.836718+0100 MyApp[9466:1237796] (c)2014-2016 Don Coleman
2019-09-06 12:53:30.837020+0100 MyApp[9466:1237796] [CDVTimer][ble] 0.666022ms
2019-09-06 12:53:30.837038+0100 MyApp[9466:1237796] [CDVTimer][TotalPluginStartup] 15.112996ms
2019-09-06 12:53:30.907136+0100 MyApp[9466:1237796] Status of CoreBluetooth central manager changed 5 (State powered up and ready (CBCentralManagerStatePoweredOn))
2019-09-06 12:53:30.907213+0100 MyApp[9466:1237796] Status of CoreBluetooth central manager changed 5 State powered up and ready (CBCentralManagerStatePoweredOn)
2019-09-06 12:53:31.975483+0100 MyApp[9466:1237796] Ionic Native: deviceready event fired after 557 ms
2019-09-06 12:53:31.994100+0100 MyApp[9466:1237796] startScan
2019-09-06 12:53:32.023843+0100 MyApp[9466:1237796] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
    };
    id = "C21ADD2E-EBB4-5C44-0F4C-256DBFF07F58";
    rssi = "-78";
}
2019-09-06 12:53:32.049945+0100 MyApp[9466:1237796] {"id":"C21ADD2E-EBB4-5C44-0F4C-256DBFF07F58","rssi":-78,"advertising":{"kCBAdvDataIsConnectable":1}}
2019-09-06 12:53:32.065435+0100 MyApp[9466:1237796] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
    };
    id = "E8DAD4A4-0E71-60FD-4F62-62193F6B10D1";
    name = iPad;
    rssi = "-34";
}
2019-09-06 12:53:32.067327+0100 MyApp[9466:1237796] {"id":"E8DAD4A4-0E71-60FD-4F62-62193F6B10D1","rssi":-34,"advertising":{"kCBAdvDataIsConnectable":1},"name":"iPad"}
2019-09-06 12:53:32.069456+0100 MyApp[9466:1237796] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 0;
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "BgABCSAC5kOfY7I4a+LPyl00byFwFsZD+jPUpFo=";
        };
    };
    id = "B45DE1D7-472E-0549-7DB3-66828F6B574D";
    rssi = "-52";
}
2019-09-06 12:53:32.079018+0100 MyApp[9466:1237796] {"id":"B45DE1D7-472E-0549-7DB3-66828F6B574D","rssi":-52,"advertising":{"kCBAdvDataIsConnectable":0,"kCBAdvDataManufacturerData":{}}}
2019-09-06 12:53:32.153237+0100 MyApp[9466:1237796] Discovered {
    advertising =     {
        kCBAdvDataIsConnectable = 1;
        kCBAdvDataLeBluetoothDeviceAddress = <003934db 4e1c7c>;
        kCBAdvDataLocalName = "[LG] webOS TV UJ651V";
        kCBAdvDataManufacturerData =         {
            CDVType = ArrayBuffer;
            data = "xAAJMwITFYA=";
        };
        kCBAdvDataServiceUUIDs =         (
            FEB9
        );
    };
    id = "B4E25B20-E93F-DE49-B421-F54DDDCCCF15";
    name = "LG webOS TV";
    rssi = "-61";
}
2019-09-06 12:53:32.154044+0100 MyApp[9466:1237796] *** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: 'Invalid type in JSON write (_NSInlineData)'
*** First throw call stack:
(0x19569a98c 0x1948739f8 0x1955a4bc0 0x19619ea48 0x1961a1950 0x195584710 0x1961a0f20 0x1961a1950 0x195584710 0x1961a0f20 0x1961a1b84 0x195583a7c 0x1961a133c 0x19605d5bc 0x19605d2e4 0x100a735c4 0x100a7ddac 0x100a73014 0x100a624cc 0x19b1b4114 0x19b1b51f0 0x19b1c3438 0x100bc36f4 0x100bc4c78 0x100bccbf4 0x100bcd8ec 0x100bd256c 0x19562c32c 0x195627264 0x1956267c0 0x19782779c 0x1c20f7c38 0x1009da5d0 0x1950ea8e0)
libc++abi.dylib: terminating with uncaught exception of type NSException
(lldb) ```

Fails here:

```
#import <UIKit/UIKit.h>

int main(int argc, char* argv[])
{
    @autoreleasepool {
        int retVal = UIApplicationMain(argc, argv, nil, @"AppDelegate");
        return retVal;
    }
}
```

don, 2019-10-17:
Duplicate of #685

closed 2019-10-17 by don