don/cordova-plugin-ble-central #692: BLE Reading Empty Buffer

LeoSaucedo, 2019-06-20:
Hello,

I've been trying to use this plugin with Ionic 4 (Cordova v 5.0.3) and debugging on Android 9.0 using ADB. I am able to scan for the device and connect to it just fine, and am able to read all of its services and characteristics.

However, when I try to read a characteristic, I am getting an empty ArrayBuffer:
```js
this.ble.read(deviceId, serviceId, characteristicId).then((data) => {
    alert(JSON.stringify(data));
}, error => {
    alert(JSON.stringify(error));
}
```

This code returns an alert that displays `{}`, meaning the success callback is triggered and an ArrayBuffer object is being returned, but it is empty. I know that this characteristic has data, because on the same device I am able to connect using [A BLE Terminal App](https://play.google.com/store/apps/details?id=de.kai_morich.serial_bluetooth_terminal&hl=en_US).

What could be happening here, or what steps should I try to take in order to debug this issue?

Thank you.

closed 2019-07-03 by LeoSaucedo

nickcom, 2020-01-25:
did you find the reason for this?

butlet, 2020-09-04:
did u find the reason for this?