don/cordova-plugin-ble-central #693: Do frequent ble.scans tend to run down batteries of the devices it returns?

markarupert, 2019-06-21:
Do frequent ble.scans tend to run down batteries of the devices it returns? I need to pull the environment fairly frequently for devices to see if they are available. Do you know if in general this type of chattiness will cause a drain on the ble devices nearby?

chris-guidry, 2019-10-14:
It shouldn't change the battery usage of the BLE device because the phone is only listening for the packets the BLE device is broadcasting. The scan isn't engaging with the device.

don, 2019-10-17:
If you scan all the time it will use more battery. Look at [autoconnect](https://github.com/don/cordova-plugin-ble-central#autoconnect) to reconnect to known devices is a more battery friendly way.

closed 2019-10-17 by don