don/cordova-plugin-ble-central #694: I can not get callback from startNotification on ios 12.3

Ariaal, 2019-07-06:
Have you encountered the same problem? and when i connected after a few time it will disconnect.
when i post data to device，and ready to startNotification but it would not get any respond.
my data is 111bytes,so i post it six times . am I right?

Ariaal, 2019-07-08:
now I found that I can not write on ios12.3

LuchoManto, 2019-07-10:
i have a similar problem, if not the same. I can connect to ble device and set readCallback. but startNotification always returns failure. I'm using an ESP32 as a peripheral with Arduino ble library. I set  the notify property in ESP32. Don't know what might be going on.

don, 2019-10-17:
startNotification should return data when the peripheral changes the characteristic value not when you write data to the peripheral.

closed 2019-10-17 by don