don/cordova-plugin-ble-central #698: java.lang.NullPointerException com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult

markarupert, 2019-08-09:
I am getting this exception reported from the field. Any ideas?

java.lang.NullPointerException
com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult

Samsung Galaxy Tab E8.0, Android 8.1

java.lang.RuntimeException: 
  at android.app.ActivityThread.deliverResults (ActivityThread.java:4519)
  at android.app.ActivityThread.handleSendResult (ActivityThread.java:4563)
  at android.app.ActivityThread.-wrap21 (Unknown Source)
  at android.app.ActivityThread$H.handleMessage (ActivityThread.java:1779)
  at android.os.Handler.dispatchMessage (Handler.java:106)
  at android.os.Looper.loop (Looper.java:164)
  at android.app.ActivityThread.main (ActivityThread.java:7000)
  at java.lang.reflect.Method.invoke (Native Method)
  at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run (RuntimeInit.java:441)
  at com.android.internal.os.ZygoteInit.main (ZygoteInit.java:1408)
  
Caused by: java.lang.NullPointerException: 
  at com.megster.cordova.ble.central.BLECentralPlugin.onRequestPermissionResult (BLECentralPlugin.java:692)
  at org.apache.cordova.CordovaInterfaceImpl.onRequestPermissionResult (CordovaInterfaceImpl.java:221)
  at org.apache.cordova.CordovaActivity.onRequestPermissionsResult (CordovaActivity.java:509)
  at android.app.Activity.dispatchRequestPermissionsResult (Activity.java:7819)
  at android.app.Activity.dispatchActivityResult (Activity.java:7642)
  at android.app.ActivityThread.deliverResults (ActivityThread.java:4515)

don, 2019-10-17:
It looks like the user denied permission. 

https://github.com/don/cordova-plugin-ble-central/blob/master/src/android/BLECentralPlugin.java#L692

Maybe the callback is null? Any idea how to reproduce? I suppose it could fail more gracefully on the null pointer.

peitschie, 2022-10-18:
Fixed in https://github.com/don/cordova-plugin-ble-central/commit/8d574ca5eb6a5e86765dda9c0a17d6c7aa1f9a1c

closed 2022-10-18 by peitschie