don/cordova-plugin-ble-central #700: iOS 13 requires new usage string

aircable, 2019-09-19:
I'm using this plugin with Ionic 5 and the app got rejected:

The app's Info.plist file should contain a **NSBluetoothAlwaysUsageDescription** key with a user-facing purpose string explaining clearly and completely why your app needs the data. Starting Spring 2019, all apps submitted to the App Store that access user data are required to include a purpose string.

I added this to plugin.xml for a quick fix:
```
<preference name="BLUETOOTH_ALWAYS_USAGE_DESCRIPTION" default=" " />
        <config-file target="*-Info.plist" parent="NSBluetoothAlwaysUsageDescription">
            <string>$BLUETOOTH_ALWAYS_USAGE_DESCRIPTION</string>
        </config-file>
```

sebj54, 2019-10-04:
Thanks to the [cordova-custom-config](https://github.com/dpa99c/cordova-custom-config#ios) plugin, we don't need to edit anything but our `config.xml` file inside the `<platform name="ios">` tag. Just add this:

```xml
<custom-config-file parent="NSBluetoothAlwaysUsageDescription" platform="ios" target="*-Info.plist">
    <string>Payzily would like to access Bluetooth to communicate with vending machines.</string>
</custom-config-file>
```

marioshtika, 2019-10-16:
Is there a plan of merging a solution for this, on the next version?

don, 2019-10-17:
@marioshtika NSBluetoothPeripheralUsageDescription is deprecated but still needed for iOS 12. There's a pull request to fix this, it should be in the next release.

don, 2019-10-17:
Fixed by #704 in version 1.2.3

closed 2019-10-17 by don

sebj54, 2019-10-18:
Thanks @don, it feels good to see the plugin updated! 👍

jptheripper, 2020-01-03:
my app is continually being rejected by the appstore because of this.  I have set 

 <custom-config-file platform="ios" target="*-Info.plist" parent="NSBluetoothPeripheralUsageDescription">
      <string>
       The application uses a serial bluetooth connection to connect to a bluetooth module on the Arduino Enoch Paintball Board for the purpose of changing configuration settings.
     </string>
    </custom-config-file>

in my config.xml file, but Apple is reporting the it says "<Appname> Would Like to Use Bluetooth" which seems like a default.  What am i doing wrong?

don, 2020-01-03:
The default usage description is an empty string. 

https://github.com/don/cordova-plugin-ble-central/blob/master/plugin.xml#L40-L47

You can specify the usage description when installing the plugin https://github.com/don/cordova-plugin-ble-central#ios

Look in appname-info.plist to see or edit the values for your app.