don/cordova-plugin-ble-central #703: Support for requestConnectionPriority API

vamshik113, 2019-10-09:
Hi Don,
For Android level 21 we have an option to set Connection interval in the form of Connection priority. If this feature could be added in the plugin it would be great...

don, 2019-10-17:
@vamshik113 I'd consider a pull request for [requestConnectionPriority](https://developer.android.com/reference/android/bluetooth/BluetoothGatt.html#requestConnectionPriority(int)). I'd imagine the implementation would look similar to [requestMtu](https://github.com/don/cordova-plugin-ble-central#requestmtu)

vamshik113, 2019-11-06:
Hi Don, I created a pull request for this feature. I have tested these changes locally using nordic sniffer and it worked. Pull request number **#714**

don, 2020-04-19:
Fixed in #714

closed 2020-04-19 by don