don/cordova-plugin-ble-central #706: BLE data received in very small chunks, fix possible?

HankLloydRight, 2019-10-20:
Hello,
  I'm currently using your plugin in a Phonegap app to receive data sent from an Arduino BLE module, but the data is being received in the onData function in very small chunks, like 10 bytes or less per call.  I have to write code to collect and parse the data stream into meaningful information. I could eliminate a lot of code if the data chunks were just a bit bigger, say 20 to 30 bytes per call to onData.  What controls this chunking?  The data is being sent, as far as I know, as a complete packet (i.e. a JSON object), but not received that way, and sometimes there's even some missing data that's never received.  But what I really like to do is just increase the chunking more than 5 to 10 bytes per call. Thanks.

don, 2019-10-21:
Can you change the peripheral so it writes a larger value to the characteristic? If not, you'll need to re-assemble the data in Javascript. The characteristic size and the peripheral control how data is sent to the phone.

HankLloydRight, 2019-10-21:
Thanks for your reply. The peripheral I'm using is a small BLE board (link here: http://www.dsdtech-global.com/2017/09/sh-m08.html), so I can't change that, unfortunately. I changed the baud rate between the Arduino and the BLE radio, but that didn't help. It's still chunked up.   Yesterday I wrote the code to listen for 120 bytes of data, and parse out the JSON objects from that (they are many small JSON objects, like **{"counter":18585}** ) so I detect the first and last one, and throw out the extra bytes at the beginning or the end.   Luckily, it's lossy and repeating data, so losing a few JSON objects in the middle won't affect anything.  And thanks for your plugin!  I've used it on a couple of projects now.

closed 2020-01-22 by don