don/cordova-plugin-ble-central #710: How to connect to HM-11 device from iOS, if only MAC address is known?

chaymankala, 2019-10-30:
I need to connect to my BLE HM11 device using iOS application. I know only the MAC Address of the device. How can I possibly identify my device in the scanresponse?

don, 2020-01-03:
iOS hides the MAC address. You need to scan for bluetooth devices, find the HM11, and connect. You can limit the scan results by filtering on the BLE service the HM11 is advertising.

closed 2020-01-03 by don

marco24690, 2023-03-30:
Hi @don so it is impossibile to use .connect() even if the macAddress is already kwnown (readed from API) in iOS?

peitschie, 2023-03-30:
Hi @marco24690 

There's no mechanism provided by iOS to get a peripheral by mac address. iOS assigns a unique peripheral UUID when a device is seen during a scan, and the UUID has no relation to the underlying MAC address which iOS deliberately hides from an application for privacy and security reasons.

On iOS, you'll always need to scan first before connecting.