don/cordova-plugin-ble-central #712: Scan not working. No error, nothing.

Xationch, 2019-10-31:
Hi,

I try to scan all the devices to start but nothing happens. No error, no device. Yet uses the Web Bluetooth I can find the devices. Any idea ?

The isEnabled function work fine.

I'm using Capacitor and Vuejs for the frontend.

iOS 13.1.3
iPhone 6s
 "cordova-plugin-ble-central": "^1.2.4",
 "@ionic-native/ble": "^5.15.1",
 "@ionic-native/core": "^5.15.1",
"@capacitor/cli": "^1.3.0",
"@capacitor/core": "^1.3.0",
"@capacitor/ios": "^1.3.0",

```javascript
<script>
import { BLE } from '@ionic-native/ble/ngx'

const serviceId = '8e450001-b2a3-f393-e0a9-e51v24dcta9d'

export default {
  data() {
    return {
      ble: BLE.prototype,
      devices: [],
    }
  },
  mounted() {
    this.scanning()
  },
  methods: {
    scanning() {
      console.log('scanning')
      this.ble.scan([], 10, function(device) {
          this.devices.push(device)
      }, () => {
          console.log('failure')
      });
    },
    onEnabled() {
      this.ble.isEnabled(() => {
        console.log("**** Bluetooth is enabled");
      }, () => {
        console.log("**** Bluetooth is *not* enabled");
      })
    },
  },
}
</script>

<template>
  <div>
    <div @click="scanning" style="background: gray; margin-top: 50px; padding: 20px;">
      START SCANNING
    </div>
    <div @click="onEnabled" style="background: blue; margin-top: 50px; padding: 20px;">
      IS BLE ENABLED?
    </div>
    <div style="color: red;">
      {{ devices.length }}
    </div>
    <pre>
      {{ devices }}
    </pre>
  </div>
</template>
```

info.plist
```
	<key>NSBluetoothAlwaysUsageDescription</key>
	<string>To connect with the Alogo Move</string>
	<key>UIBackgroundModes</key>
	<array>
		<string>bluetooth-central</string>
	</array>
```

closed 2020-01-09 by Xationch

Pratikshakhandagale, 2020-01-22:
@Xationch You found any solution ?

Xationch, 2020-01-22:
@Pratikshakhandagale Yes, you need to use javascript subscribe() :

```javascript
    import { BLE } from '@ionic-native/ble

    BLE.scan([], 10)
      .subscribe(device => {
        console.log('new device', device)
      })
```