don/cordova-plugin-ble-central #716: Read characteristic failed

chegewara, 2019-11-10:
Hi,
i am trying to read characteristic from peripheral, but always promise returns fail:
```
async ReadSettings() {
    console.log('read settings');
    if (this.ble.isConnected(this.myNavService.device.id)) {
      console.log(this.myNavService.device.id, this.myNavService.writeService, this.myNavService.settingsCharacteristic);
    }

    return this.ble.read(this.myNavService.device.id, this.myNavService.writeService, '6e400002-b5a3-f393-e0a9-e50e24dcca9e')
    .then((data) => {
      const json = JSON.parse(data);
      console.log(json);
      return json;
    }).catch(reason => {
      console.log('cant read settings');
      console.log((reason));
      throw reason;
    });
  }
```
logs:
```
[ng] [console.log]: "read settings"
[ng] [console.log]: "30:AE:A4:43:A1:72" "6e400001-b5a3-f393-e0a9-e50e24dcca9e" "6e400002-b5a3-f393-e0a9-e50e24dcca9e"
[ng] [console.log]: "cant read settings"
[ng] [console.log]: {}
```

Yes, my peripheral is configure with READ property on that characteristic and it can be read from android smartphone with nRF connect.
I am using ionic 4 with angular and tested on 2 samsung smartphones (S5 and S9+) with different peripheral (esp32 and adroid smartphone with nRF connect).

monkeytronics, 2019-12-01:
Hey. Could it be the Array Buffer tripping up on the JSON.parse. Maybe try: 

```
// ASCII only
bytesToString(buffer) {
      return String.fromCharCode.apply(null, new Uint8Array(buffer));
  }
```

Copied from the ionic native plugin page. I was getting apparently empty objects too til I did this.

chegewara, 2019-12-01:
Thanks for answer. I am currently using this plugin and it works great:
https://ionicframework.com/docs/native/bluetooth-le

closed 2020-01-22 by don