don/cordova-plugin-ble-central #717: iOS State Preservation and Restoration

adelin-mcbsoft, 2019-11-12:
Hi,

Long-story-short: BLE _background-mode_ works fine as long as `UIBackgroundModes` has `bluetooth-central` enabled in `Info.plist`. The behavior is as follows:

Once the app. goes from foreground to background (to _'suspended'_ state), every time there is BLE traffic, iOS wakes up the app. in background and the app. behaves according to its code.

But, after some time (usually few hours of non-usage), the app. gets killed by iOS, so even if there's BLE traffic, nothing happens anymore as the app. is dead.

However, iOS allows a _"State Preservation and Restoration"_ behavior, so whenever there's a previously connected BLE device that wants to interact with your app., iOS can relunch the app. in background with a callback to your code, so it'll do what it needs to do.

Read more about [here](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW5), on Apple's Developers Docs .

I've seen _Bluetooth LE plugin_ developed by _Rand Dusing_ ([cordova-plugin-bluetoothle](https://github.com/randdusing/cordova-plugin-bluetoothle/)) has a similar [open issue](https://github.com/randdusing/cordova-plugin-bluetoothle/issues/433), maybe it can shed some light.

I don't really know if Android has a similar implementation or not, but for iOS is a feature that would be very useful for long-term BLE communications between phone and other devices.

Thanks,
Best,
A.

isaiahtaylor, 2019-12-27:
I second this feature request! That would be very helpful functionality.

don, 2020-01-22:
This sounds like a good enhancement to the plugin.

peitschie, 2022-02-07:
This feature has now been implemented.

To turn it on, a few steps are needed:
1. set the `BLUETOOTH_RESTORE_STATE` variable to `true` (e.g., `cordova plugin add cordova-plugin-ble-central --variable BLUTOOTH_RESTORE_STATE=true`)
2. Add the a section to the `config.xml` to allow the `bluetooth-central` background mode (as per the README):
```
<platform name="ios">
    <config-file parent="UIBackgroundModes" target="*-Info.plist">
        <array>
            <string>bluetooth-central</string>
        </array>
    </config-file>
</platform>
```

It's important to note that iOS only gives roughly 10s of processing time after waking up due to a Bluetooth event, during which time an implementing application must re-establish any relevant BLE comms. E.g., if an app is woken up due to a scan picking up a BLE device, the app must try and initiate a connection within 10s or risk being suspended again.

While the BLE device and the phone are in active BLE comms, the app will generally be allowed to process.

There's quite a few caveats and things noted in the linked issue at https://github.com/randdusing/cordova-plugin-bluetoothle/issues/433 from cordova-plugin-bluetoothle that are worth reading to learn the details of the system.

adelin-mcbsoft, 2022-02-07:
Hi [peitschie](https://github.com/peitschie).

Thank you for this major improvement, it really comes up very handy.

Question: when the app. wakes up due to restoration, does it trigger any particular event, so we could know which of the previously connected Bluetooth devices triggered the wake-up?

I haven't tested your release yet, but I remember from my previous researches that you have to restore the connection back by using the previously passed `restoreKey`, otherwise it will indeed time out.
See my last two posts from [here](https://github.com/randdusing/cordova-plugin-bluetoothle/issues/433#issuecomment-554202805), where I described the detailed process.

Therefore, by having some connection information from a "wake-up event", you could actually reconnect to the device, without having to loop through all the past connections.

Thanks!

peitschie, 2022-02-07:
Hi @adelin-mcbsoft 

Thanks for that detailed write-up. It was very informative starting point I used when trying to get this setup going 🙂

The change I've made to the plugin always uses a hardcoded `restoreKey` when the configuration option is active, which simply rehydrates the `centralManager` back to it's prior state of scanning/connecting. From the plugin's perspective, it doesn't (currently) even try and reconnect or reload these itself, relying on the cordova JS app to handle the heavy lifting.

For my own case here, the workflow on app launch is always:
1. Check to see if a companion BLE peripheral is configured
2. If none configured, start scanning for relevant device
3. If one is configured, immediately `connect` again

If this has come via a restore, step 3 happily resumes with no error as the peripheral UUID is already known by the central. As iOS already has established a connection (hence the wake-up event), the connect request completes immediately, and the app can continue on happily.

For this same reason, I haven't gone to the effort of supporting a wake-up event. From the application's perspective, it's either suddenly doing something after being idle for a while (suspend) or is being launched completely cleanly (user launched or restore after iOS termination).

I struggled to figure out how to "rehydrate" not just the central manager but also all the required hooks in a client application when a restore occurs. In the end, it seems the most robust app was one which just went through the same steps it did as if the user launched it cleanly.

But.... I'm happy to hear more details about what issue you're trying to solve though, in case there's a straightforward increment that can help you address your need also.

peitschie, 2022-02-07:
Right, so reading your analysis there in more detail:

1. iOS relaunches the app when a BLE scan reports a device
2. iOS relaunches the app when a BLE notification arrives

Unlike the connect/disconnect events which don't really lose any info as an app can relatively easily determine this state change itself, scan events and notifications are transient and will likely arrive and be discarded before the app finishes initialising.

Agree, those cases are not yet covered here 🤔

peitschie, 2022-02-07:
It's not really possible to make a callback-based solution with the current approach, because the `centralManager` is initialised before any callbacks from JS can be handed to the plugin.

@adelin-mcbsoft would it be sufficient to simply store that restoration dictionary and allow it to be retrieved later? E.g., have a `retrieveRestorationState` method exposed which can be used to fetch the `dict` argument passed into `(void)centralManager:(CBCentralManager *)central willRestoreState:(NSDictionary *)dict`?

adelin-mcbsoft, 2022-02-07:
So, I will try to retest the behavior ASAP before jumping to conclusions, but these things just came into my mind:
(I will number them, so we can easily keep track of):

**1. RestoreKey**
So, if I remember correctly, the restore key should be an unique identifier per app.
Having it hardcoded would not be a great idea, since you could have two different apps using the same plugin.
In that situation, whenever a restore key will occur, iOS will fail to relaunch the correct app.

**2. Wake-up process**
If I understood correctly, using the current implemented functionality, whenever the iOS device will be around a previously Bluetooth device, it will just relaunch the app. in the background, and JS should start scanning for devices and when it finds a suitable one, just (re)connect to it.

I'm an _"optimization-guy"_ and I foresee two problems here:

**2.1** First, the lack of an event does not offer the possibility for the developer to prioritize (just) the BLE workflow.

If an event would be available, in the app. logic we could know that the app was launched due to BLE coming to life, and prioritize (just) the BLE workflow instead of loading the UI, making connections to APIs and whatever else _"normal"_ logic we would have whenever an user opens the app normally.

Knowing that the app alive in the background, we allow us to address just the BLE part until the app. comes to foreground. A lot of time and battery would be saved in this scenario.

**2.2** Further more, as we don't have any connection information from any wake-up event, precious time (and battery) is wasted when trying to rescan, stop scan and connect to a device.
In my previous tests, with some devices it could take even 2-5 seconds to do so.

If we'd have the connection information from the `centralManager` (for example, which UUID triggered the BLE and app opening), we could directly **reconnect** to that device without the hassle of re-scaning and so on.

--

_Personal note:_
_Hope my input helps and hope I'm not 'wanting' too much._
_It's just that there are only these two BLE plugins available, this one in particular was missing this feature entirely (reason why I have to give it up), and the other one lacks the features described above._

_That's why I'm so eager to get at least one of them work flawlessly and I'm happy to invest as much time as it takes to do so.
Too pity I'm not an iOS Dev, but if I can help with anything else, just let me know - I'm so happy there's finally someone out there who could help with this._

**3. Later edit** (as I read your last 2 replies before I wrote all these):
Yes, if it's not possible to pass the information from `centralManager` as it gets initialized before JS, even storing the info somewhere and retrieving after JS's `ready()` event would be great. Anything will work.

Looking forward to hearing from you,
Thanks,
A.

peitschie, 2022-02-07:
>  So, if I remember correctly, the restore key should be an unique identifier per app.

Yeah, the [documentation](https://developer.apple.com/library/archive/documentation/NetworkingInternetWeb/Conceptual/CoreBluetooth_concepts/CoreBluetoothBackgroundProcessingForIOSApps/PerformingTasksWhileYourAppIsInTheBackground.html#//apple_ref/doc/uid/TP40013257-CH7-SW10) is unclear on this point. Usually, UUID or GUID indicate a universally unique identifier, so the use of UID here is a bit confusing.

I'll switch to the app bundle identifier with an option to override via the preference key. This will hopefully give sensible defaults to the system.

I'll ponder a bit how best to expose the event data you've asked for. It certainly would be nice to have this able to restore all these things again...

peitschie, 2022-02-08:
@adelin-mcbsoft I've now added a `restoredBluetoothState` function that returns an object similar to this:

```
2022-02-08 17:16:12.616948+1100 App[972:301238] restored BLE state was: {
 "scanServiceUUIDs": [
  "54FA2EC5-FAE7-4AD9-9F8E-AB5267702AAB"
 ],
 "peripherals": [
  {
   "characteristics": [
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2ABD",
     "service": "1825"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2ABE",
     "service": "1825"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2ABF",
     "service": "1825"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2AC0",
     "service": "1825"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2AC1",
     "service": "1825"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2AC2",
     "service": "1825"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2AC3",
     "service": "1825"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2AC4",
     "service": "1825"
    },
    {
     "properties": [
      "Write",
      "Indicate"
     ],
     "isNotifying": false,
     "characteristic": "2AC5",
     "service": "1825"
    },
    {
     "properties": [
      "Write",
      "Indicate"
     ],
     "isNotifying": false,
     "characteristic": "2AC6",
     "service": "1825"
    },
    {
     "properties": [
      "Read",
      "Write"
     ],
     "isNotifying": false,
     "characteristic": "2AC7",
     "service": "1825"
    },
    {
     "properties": [
      "Indicate"
     ],
     "isNotifying": false,
     "characteristic": "2AC8",
     "service": "1825"
    },
    {
     "properties": [
      "Read",
      "Write",
      "Notify"
     ],
     "isNotifying": false,
     "characteristic": "2A2B",
     "service": "1805"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2A29",
     "service": "180A"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2A24",
     "service": "180A"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2A25",
     "service": "180A"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2A27",
     "service": "180A"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2A26",
     "service": "180A"
    },
    {
     "properties": [
      "Read"
     ],
     "isNotifying": false,
     "characteristic": "2A28",
     "service": "180A"
    }
   ],
   "id": "8B543A80-1E90-C6D5-FFD8-DFB7F34DCE07",
   "state": "connected",
   "services": [
    "1825",
    "1805",
    "180A"
   ],
   "name": "MPY NIMBLE"
  }
 ],
 "scanOptions": {
  "kCBScanOptionAllowDuplicates": true
 }
}
```

There is still an unaddressed issue in that if a notification wakes the app back up, the notification will be lost. But... this one is harder to solve without introducing some kind of unintentional memory leak or similar, as notifications can be lost up until the point that the app completes reconnecting and resubscribing to anything.

peitschie, 2022-02-15:
I'm closing this issue out as mostly implemented, and raised an issue for the remaining corner case. What's landed is adequate for several use cases. I'll need to ponder the remaining corner cases more to figure out if there's a nice way to address them in a future release.

Thanks for the discussion @adelin-mcbsoft ! If you do hit any issues with this, obviously feel free to raise additional issues and I'll cover them as best I can 🙂

closed 2022-02-15 by peitschie