don/cordova-plugin-ble-central #719: No code for bluetoothSerial.clearDeviceDiscoveredListener()

sammy00747, 2019-11-18:
Hi!

Seems like although the usage for bluetoothSerial.clearDeviceDiscoveredListener() is present in the API docs, there seems to be no existing function definition for the same.

A quick solution is appreciated.

Thanks in advance.

sammy00747, 2019-11-18:
I had to 'literally' write these lines in /node_modules/@ionic-native/bluetooth-serial/ngx/index.d.ts, after "setDeviceDiscoveredListener(): Observable<any>;":

/**
     * Clears notify callback function registered with setDeviceDiscoveredListener.
     * @returns {void}
     */
    clearDeviceDiscoveredListener(): void;

closing the issue, and waiting for these changes to be implemented.

sammy00747, 2019-11-18:
Sorry to say but this issue is still not solved. Please look into it.

don, 2020-01-03:
@sammy00747 it sounds like that code is missing from the Ionic Native wrapper, not this plugin.

sammy00747, 2020-01-08:
Thanks for your response @don!! I would look into this.

closed 2020-01-08 by sammy00747