don/cordova-plugin-ble-central #723: Printer write response not coming

kalemayur8, 2019-12-04:
Hi Don,

Thanks for such good plugin, I have been using this since some time. I am facing one issue during write operation on printer.

does write success function returns response also ? if no then what is the way to collect printer response.

I tried bte.startNotification before bte.write which collects data byte by byte and difficult to understand. 

Appreciate your quick help

natestone, 2019-12-13:
You need to convert the byte array to a string to read it.  Otherwise, it sounds like you're on the right track.  Here's a sample snippet.

`  bytesToString(buffer) {
    return String.fromCharCode.apply(null, new Uint8Array(buffer));
  }`

kalemayur8, 2019-12-15:
We are sending multiple lines of data to printer in Synchronously , the notification method take huge time to receive this data byte by byte and hard to understand which response is of which write operation. Is there any better way ?

don, 2020-01-03:
I don't know. This plugin is just providing the communication layer, you'll need to work out the device API with the printer manufacturer. If you can find and existing app that works with the printer, use Wireshark and a bluetooth sniffer to record the traffic to see what they are doing.

closed 2020-01-03 by don