don/cordova-plugin-ble-central #724: Android: startScanWithOptions stops scanning after 30 minutes

barakataboujreich, 2019-12-17:
hello,
i am working on an app that keep scanning for our ble devices and send their advertisement data to our server. i noticed that startScanWithOptions stop returning any device exactly at 30 minutes mark on some of our android devices. after some research i was able  to find the cause of the problem:
**Android 7.0 introduced a BLE scan timeout, where any scan running for 30 minutes or more is effectively stopped automatically and only resumed "opportunistically" which essentially means that if another process does a scan, it can get the results as well.**

and indeed the android devices that stop scanning after 30 minutes run on android 8.0.0,  8.1.0  and 9, the devices that keep scanning run on android 7.0.0 and 6.0.1

is their any solution for this??
i could set an interval that stop and start scanning every 30 minutes, but this is only a work around

don, 2020-01-03:
Continuously scanning uses a lot of battery. If you really need to do this, use the interval. Perhaps you can find another solution like using auto connect.

don, 2020-04-19:
I think that using autoConnect or just resetting the scan timer after 29 minutes is the way to go.

closed 2020-04-19 by don

jplajpla23, 2021-04-09:
Hi @don, how can I reset the timer. only need stop scan and start again?
my app is used to read beacon broadcast withtout no time limit

barakataboujreich, 2021-04-09:
there is no infinite time limit. it is forced to stop after 30 minutes. 
what i did is i set and interval that stop scanning and start scanning every 29 minutes, that allowed me to constantly scan all the time. in my case, i missed couple of scans from vehicles going 100 km/h, but can't do nothing about it