don/cordova-plugin-ble-central #725: Photos Capturing issue on iOS using Pulse Camera Remote in Canon DSLR Camera

mohit-testframework, 2019-12-24:
Hi Don and All,

Thanks for this nice plugin. I am using this plugin for connecting "Pulse Camera Remote" with the app and capturing photos, videos, and doing camera settings in Canon DSLR Camera. Please check the Pulse camera remote link : (https://alpinelaboratories.com/products/pulse-camera-remote).

I am using this plugin in the Ionic app (version 1.x). I would like to request to check this video to understand how the app's functionality working (https://www.youtube.com/watch?v=PrP5g6NH2lA).

**Issue:** 
The code is working fine when Pulse Remote is just connected and we capture photo then photo capturing is working fine, but after some time automatically photo capturing feature does not work. I don't know what is the issue occurring; The Pulse, Camera, and mobile app are connected but still, the photo is not capturing. Please check the BLE.write code below:

          var captureData = [0x00, 0x01, 0x11, 0x11];
         console.log('captureData : ' + JSON.stringify(captureData));
         //Value of above console is =>  captureData : [0,1,17,17] 

          var buff = new Uint8Array(captureData);
         console.log('buff : ' + JSON.stringify(buff)); 
          //Value of above console is =>  buff : {"0":0,"1":1,"2":17,"3":17}

          ble.write(device.id, "1901", "3006", buff.buffer, function(response) {
           console.log('BLE writeSuccess  : ' + response);
            //Value of above console is returning "null". 
          deferred.resolve(response);
        }, function(error) {
           console.log('BLE writeError  : ' + error);
          deferred.reject(error);
        });

Also, the response of ble.write method returns "null".

1. How can I fix the photo capturing issue?
2. How can I print the ble.write's success response?

Please help me on this, Thanks in Advance.

Thanks
Mohit

mohit-testframework, 2019-12-27:
Hello All,
Please help me with this.
Thanks

don, 2020-01-03:
I think this is a problem with your hardware rather than the plugin. The [product page](https://alpinelaboratories.com/products/pulse-camera-remote) you linked to said "There are compatibility issues with the Pulse App with Android 9 and 10 and iOS 12.4 and later. "

When you call write, you're not going to get data back. It will indicate success or failure. (I can't recall what the Ionic 1 bindings do, probably return a promise.) If you need a response from the hardware after writing, this typically comes back to the phone via a characteristic notification.

I'd try running your app in a debugger to track down the issue. If it's really a Bluetooth device issue, you might have better luck capturing Bluetooth traffic in Wireshark with something like the Nordic nRF development kit or the Adafruit Bluefruit Sniffer https://www.adafruit.com/product/2269

closed 2020-01-22 by don