don/cordova-plugin-ble-central #726: autoConnect doesn't work after 24hrs on iOS

isaiahtaylor, 2020-01-01:
I have an app that basically needs to gather data from a peripheral whenever it is in range and available to connect, often without any user interaction for weeks at a time.

Using autoConnect, I've gotten the device to connect and disconnect multiple times by walking near the peripheral and away from it while the app has been in the background for a couple hours. However, **the next morning I had no luck getting it to connect without opening the app**.

I'm aware that autoConnect is not meant to address "quit" events (user swiping up), but it seems that it should work for long term app sleep, and wake the app up?

Thanks for the help.

don, 2020-01-03:
It should work over longer periods of time. Is it possible the UUID on the peripheral changed? Maybe something changed on iOS 13. Make sure you have [background notifications enabled](https://github.com/don/cordova-plugin-ble-central#background-notifications-on-ios). It's possible that it takes a long time (minutes) to reconnect. Check Apple's iOS documentation to see the expected behavior. If you can get some log messages about what's happening when the reconnect fails, that would be helpful.

isaiahtaylor, 2020-01-04:
Thanks for getting back. I will try to recreate the issue and see if waiting longer resolves it.

JesperBalslev, 2020-01-08:
Hey @don thanks for the good work.
I have an issue that might be related (let me know if i should create another topic).

With autoConnect, every now and then, my devices are unable to connect, i don't receive any success or error callbacks, and my only way to fix this, is by enabling flightmode (clearing cache?), and wait for 5-10 seconds, disable flightmode and then restarting the app.  

If i have this issue, even changing the code to not using autoConnect will not help. Only doing the flightmode "trick", or reinstalling the app fixes this.

Not sure if my device service changed characteristics somehow, or another issue happened with the cache.

I tried playing around with refreshDeviceCache(), but i suspect i need to be connected to use that, since it fails all the time.

Any help would be appreciated.

TobiasFP, 2020-01-27:
@isaiahtaylor Could you resolve it by waiting longer?

isaiahtaylor, 2020-01-27:
@TobiasFP I'll give it a shot tonight and let you know if I can sit there for 10 mins and connect in the  morning.

TobiasFP, 2020-01-27:
Thank you man, much appreciated.

isaiahtaylor, 2020-02-12:
@TobiasFP I was not able to get this working. After talking to an iOS BLE expert, I believe there's a further state of "sleep" that iOS enters after a certain amount of time which has increased memory pressure. I think Cordova is unable to stay within those limits, and so fails to come up when it is woken up by the BLE event.

TobiasFP, 2020-02-13:
My connection has been working fine for 16 hours once, but i haven't tested further.

isaiahtaylor, 2020-03-11:
Hello all, just giving a quick update on this.

As far as we can tell, a constant BLE connection can survive longer than 12 hours when there is data passing back and forth constantly. However, if there is longer than a ~12 hour gap (I'm not sure what time iOS decides to put the app into a lower state of suspension) between any BLE events, the app fails to wake up. We surmise that this is because the window iOS gives the app to perform its BLE actions and go back to sleep includes too little memory to spin up Cordova. So the app comes up but Cordova fails to start in it and the app goes back to sleep. This seemed to be confirmed by the fact that it would *sometimes* work. Once in a blue moon, the memory pressure is not as high when the event comes through and iOS gives the app enough leeway to start up.

I think this is a Cordova issue, not an issue with this plugin. This constraint, combined with lack of iOS preservation and restoration, means that we have switched to Flutter. I'm fine with this issue being closed and another opened with Cordova.

don, 2020-04-19:
@isaiahtaylor thanks for the update

closed 2020-04-19 by don