don/cordova-plugin-ble-central #727: ble.isEnabled() takes 0 arguments?

HassanMehdi17, 2020-01-03:
Hello,

Am i missing something here. the docs say the function takes 2 callback methods. but in code it doesnt allow me to put anything between the parenthesis? are the docs out dated? how would i get a success/failure messages?

thank you in advance

don, 2020-01-03:
You get success or failure in the callback functions. The example in the docs is correct. https://github.com/don/cordova-plugin-ble-central#isenabled

HassanMehdi17, 2020-01-03:
[image]

can i get an explanation for this then?

don, 2020-01-03:
According to the linter message, `this.ble` is not the same as the code referenced in the documentation. It's possible you're using a different BLE library or the promises version of the API.

I recommend starting simple and writing some throwaway code using plain javascript to get an idea of how the API works. While you're learning how to use the plugin and API, Stackoverflow is a better place to post this type of issue.

HassanMehdi17, 2020-01-03:
thank you for the information,

from the ionic framework website their BLE page has a link to this github:
[https://ionicframework.com/docs/native/ble](url)

and followed the instruction in their usage section after installing

I'm importing:
`import { BLE } from '@ionic-native/ble/ngx';`

and in the constructor:
`  constructor(
    public alertController: AlertController, 
    private modalCtrl: ModalController, 
    private ble: BLE,
    private ngZone: NgZone) { }`

how could i make it not the promises version of the api?

don, 2020-01-03:
That's where the confusion is. Ionic provides a TypeScript wrapper around this plugin, so the Ionic API will be different that this plugins API. Rather than passing callbacks to the functions with Ionic Angular most functions will return Promises or subscriptions to Observables. (I'm guessing the React stuff is similar.)

I have some old Ionic 3 examples that might help https://github.com/don/ionic-ble-examples

closed 2020-01-03 by don

HassanMehdi17, 2020-01-03:
ohh that makes a bit more sense. thank you