don/cordova-plugin-ble-central #728: getting different name for peripheral on iOS and android

HassanMehdi17, 2020-01-05:
i think i understand that there is a different between GATT data and GAP data. i used the eddystone app to change the name of my BLE device to "TBeacon" from "pBeacon_n". on iOS it always shows as TBeacon. but on android for 2 minutes after turning on, it shows as pBeacon_n in a scan. and after 2 minutes scanning shows TBeacon on android.
 i tried decoding the advertising data and used the SERVICE_DATA_KEY 0x09 or 0x08 from the technique that Don has posted here before, but it still only shows the pBeacon_n. is there a way to access the 0x1800 name which should have the name that i'm looking for?

HassanMehdi17, 2020-01-17:
no reply?

droghio, 2020-01-17:
Simple answer, if you connect you can read the name service like any other
if it is available on the device.

Long answer...
Phone caches names heavily. A device name can be transmitted in the
advertisement and/or stored in the GATT database. They can also be
different, and the phone might cache an old entry from either of these.
However the library will always give you the latest data if you directly
parse the advertisement response or query the name service.

From what I recall Eddystones are strictly beacons so you don't usually
access the GATT database directly but use the extra fields included in the
advertisement to estimate distance. So I assume for your purpose you should
be looking at the advertisement like you started out.

My guess is the Eddystone is still broadcasting the old name in an
advertisement. A quick test would be to use LightBle on the phone to dump
the raw advertisement packet. It might help if you also post the
advertisement data received by the scan.

don, 2020-01-22:
@HassanMehdi17, Like @droghio said, you're fighting caching sometimes. Reading from the advertising data is good. You might need to do this multiple times. There's the scan and scan response which can contain different data. On Android you can read 0x1800, but I think iOS hides this from you.

closed 2020-01-22 by don