don/cordova-plugin-ble-central #731: Getting phantom "stop notification" message but complete callback is never called

isaiahtaylor, 2020-01-11:
Hello, after registering for the notification callback, I sometimes get a phantom message in the logs that says "stop notification," after which I no longer get notification callbacks. The "error" and "complete" callbacks are also never called.

```ts
this.ble
  .startNotification(this.periph.id, notif.service, notif.characteristic)
  .subscribe(
    data => this.receiveRelay(data),
    error => {
      console.log('error');
      console.log(error);
    },
    () => {
      console.log("ended");
    }
);
```

Logs:

```
received
[the data I got]
stop notification
```

After this "stop notification" message, I don't get any notifications. I assume there is an error happening somewhere, but I don't get any error logs.

don, 2020-01-12:
Can you try and do the same thing using callbacks instead of the ionic wrapper?

https://github.com/don/cordova-plugin-ble-central#startnotification

isaiahtaylor, 2020-01-15:
I ended up finding a bug downstream of one of my notification callbacks. I still don't know why there is no error shown or why that error propagated all the way to cancelling notifications, but that is out of scope of this issue. Feel free to re-open.

closed 2020-01-15 by isaiahtaylor