don/cordova-plugin-ble-central #733: Peripheral XX:XX:XX:XX:XX:XX not found attempting to disconnect

markarupert, 2020-01-22:
My connection is getting into a state where the notifications continue to come in but the device will fail on writes with:

ble.write fail Peripheral not found

I attempt a disconnect and reconnnect. However, when I attempt to disconnect I get the error again:

Peripheral XX:XX:XX:XX:XX:XX not found

Is there any other way to force a disconnect programatically?

don, 2020-01-22:
Which platform? You should use the callback from connect to know when the device disconnects. It's possible it disconnected before you call write.

If the peripheral isn't advertising, you won't be able to connect. Try rebooting the peripheral. It's possible the peripheral is connected to your device but the plugin can't see it. Try toggling bluetooth on the phone.

markarupert, 2020-01-22:
Thanks for the quick response Don. 

I have implemented that callback from the connect and it is not getting called in this case.

The device sends a message every second and I am continually getting that data through the notify even when the write fails so I know that it is still connected.

Rebooting the device or toggling bluetooth on the phone works on the bench but the devices are in the field and I am attempting to handle the disconnect reconnect in the background without involving the user.

don, 2020-01-22:
If you're still connected and getting notifications, it sounds like you're connected. Can you check the case on the mac address? Other than that, maybe debug setting a breakpoint in the Android code to see why the peripheral lookup fails.

https://github.com/don/cordova-plugin-ble-central/blob/7295e9e5d591e29db3dfbd23dac19b8240482619/src/android/BLECentralPlugin.java#L519-L524

markarupert, 2020-02-07:
I will have to try an recreate it on the bench this is all happening in the field. The only solution is to cycle the app. Is there any way to restart/reload the ble plugin programatically like it does at the App startup. Just drop all connections and start over?

don, 2020-04-19:
>  Is there any way to restart/reload the ble plugin programatically like it does at the App startup. Just drop all connections and start over?
No I think you'd need to cycle through that connections and close them.

closed 2020-04-19 by don