don/cordova-plugin-ble-central #737: java.lang.ClassNotFoundException: com.megster.cordova.ble.central.BLECentralPlugin

vidarmikaelsson, 2020-02-05:
Hi, while running an Ionic Application (5.4.16) on a real Android device (Q: 10) I get the following Error:

`V/Capacitor/Plugin: To native (Cordova plugin): callbackId: BLE1053240627, service: BLE, action: scan, actionArgs: [[],5]
W/System.err: java.lang.ClassNotFoundException: com.megster.cordova.ble.central.BLECentralPlugin
        at java.lang.Class.classForName(Native Method)
        at java.lang.Class.forName(Class.java:454)
        at java.lang.Class.forName(Class.java:379)
        at org.apache.cordova.PluginManager.instantiatePlugin(PluginManager.java:489)
        at org.apache.cordova.PluginManager.getPlugin(PluginManager.java:169)
        at org.apache.cordova.PluginManager.exec(PluginManager.java:122)
        at com.getcapacitor.MessageHandler.callCordovaPluginMethod(MessageHandler.java:70)
        at com.getcapacitor.MessageHandler.postMessage(MessageHandler.java:46)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:336)
        at android.os.Looper.loop(Looper.java:197)
        at android.os.HandlerThread.run(HandlerThread.java:67)
W/System.err: Caused by: java.lang.ClassNotFoundException: Didn't find class "com.megster.cordova.ble.central.BLECentralPlugin" on path: DexPathList[[zip file "/data/app/io.bwt.freddie-D39OMkuCqHe0weOJfeA49g==/base.apk"],nativeLibraryDirectories=[/data/app/io.bwt.freddie-D39OMkuCqHe0weOJfeA49g==/lib/arm64, /system/lib64]]
        at dalvik.system.BaseDexClassLoader.findClass(BaseDexClassLoader.java:196)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:379)
        at java.lang.ClassLoader.loadClass(ClassLoader.java:312)
    	... 12 more
I/System.out: Error adding plugin com.megster.cordova.ble.central.BLECentralPlugin.
E/Capacitor: Post message error:
    java.lang.NullPointerException: Attempt to invoke virtual method 'void org.apache.cordova.CordovaPlugin.privateInitialize(java.lang.String, org.apache.cordova.CordovaInterface, org.apache.cordova.CordovaWebView, org.apache.cordova.CordovaPreferences)' on a null object reference
        at org.apache.cordova.PluginManager.getPlugin(PluginManager.java:171)
        at org.apache.cordova.PluginManager.exec(PluginManager.java:122)
        at com.getcapacitor.MessageHandler.callCordovaPluginMethod(MessageHandler.java:70)
        at com.getcapacitor.MessageHandler.postMessage(MessageHandler.java:46)
        at android.os.MessageQueue.nativePollOnce(Native Method)
        at android.os.MessageQueue.next(MessageQueue.java:336)
        at android.os.Looper.loop(Looper.java:197)
        at android.os.HandlerThread.run(HandlerThread.java:67)`

I followed the example of https://ionicframework.com/docs/native/ble to install the plugin:

`ionic cordova plugin add cordova-plugin-ble-central
npm install @ionic-native/ble`

Component in Ionic/Angular App:

`import { Component, NgZone } from '@angular/core';
import { BLE } from '@ionic-native/ble/ngx';

@Component({
  selector: 'app-ble-connect',
  templateUrl: './ble-connect.page.html',
  styleUrls: ['./ble-connect.page.scss'],
  providers: [BLE]
})
export class BleConnectPage implements OnInit {
  devices: any[] = [];

  constructor(private ble: BLE, private ngZone: NgZone) { }

  Scan() {
    this.devices = [];
    this.ble.scan([], 5)
    .subscribe(device => this.addDiscoveredDevice(device));
  }

  addDiscoveredDevice(device) {
    this.ngZone.run(() => {
      console.log('Discovered' + JSON.stringify(device, null, 2));
      this.devices.push(device);
      console.log(device);
    });
  }

}
`

The app is the build using
`ionic capacitor run android`

Any pointer into the correct direction would be greatly appreciated.

vidarmikaelsson, 2020-02-08:
Completely reinstalled ionic app and did all steps again. Problem was gone...

closed 2020-02-08 by vidarmikaelsson

lukethacoder, 2020-02-29:
Had a similar issue. Rebuilding the android app fixes the problem.

Must have not been pulling through new packages added in package.json through to Gradle properly 🤔