don/cordova-plugin-ble-central #738: BLE startNotification should notify any state change of the bluetooth device even when phone is locked/sleep mode

meghajayakumar, 2020-02-07:
App is an ionic4 capacitor app which uses few cordova plugins as well. I have a timer feature associated with a bluetooth device that has to keep running even when the phone goes to sleep or in locked mode. Using the cordova-background-mode plugin the app works in the background, but as soon as phone is locked or in sleep the timer stops working. When the screen resumes, the timer continues from where it stopped.

I cannot use the platform pause and resume events in my case as just taking the difference and adding up with the time will provide wrong values with my use case.

This app is used with the height adjusting table connected via bluetooth, where a user can use the app to adjust the height and also track the time when he is in standing position. (The timer starts running when the table reaches a certain height - achieved using the startNotification method of ble). The height adjusting table also has button to control its height.

Use case (Why I cannot use the platform resume and pause events along with timestamps) :

1. Desk is in standing position. Timer starts running.
2. I lock my phone (timestamp created)
3. I go in sitting position for 1h (The table also has buttons to control the height, so this is referred to when the user uses that to reduce the height and is in sitting position, but the phone is still locked/sleep and is not notified about the change)
4. I go back to standing position for 20min
5. I unlock my phone. (Second time stamp created)
6. Now it will show 1:20min standing time even though I stood only 20min

How can I make wake the phone or notify when the ble device state changes even when the phone is locked? I tried by including the script inside the startNotification method which would create a timestamp in the DB when the height is adjusted. But this doesn't work if the phone is locked/sleep. 

Any state change of the bluetooth device should be notified even when the phone is locked so that the timer would show correct values of standing time. Is there any other method with which I can achieve the same?

don, 2020-04-19:
This might be related to #717 and https://github.com/don/cordova-plugin-ble-central/issues/726#issuecomment-597781600

peitschie, 2021-11-16:
@meghajayakumar are you able to clarify whether this is Android or iOS?

peitschie, 2022-02-15:
As of 1.4.2, it's possible to enable background processing modes on iOS to help with this situation: https://github.com/don/cordova-plugin-ble-central/pull/888

However, no progress has been made on the Android front.

peitschie, 2023-03-16:
The most likely cause of this issue on Android is due to the webview timers in Cordova being paused when the app is not visible. A plugin such as https://bitbucket.org/TheBosZ/cordova-plugin-run-in-background, specifically using the `cordova.plugins.backgroundMode.disableWebViewOptimizations()` method, which inhibits the timer pause.

Other than this, the other likely cause is Android's own power management destroying the activity.

Fundamentally though, this plugin will work correctly in the described situations to the best of my knowledge.

closed 2023-03-16 by peitschie