don/cordova-plugin-ble-central #740: Can't Figure out How to Add Plugin to PhoneGap

ZebNZ, 2020-02-12:
Hi all,

I am new to PhoneGap and am trying to find how to install the BLE Plugin!

if I run this command in the CLI:
`phonegap plugin add cordova-plugin-ble-central` 
I get this error:
```
'phonegap' is not recognized as an internal or external command,
operable program or batch file.
```

How am I meant to install the plugin?

Thanks,

Zeb

ZebNZ, 2020-02-17:
......also do you need Node.js installed

don, 2020-04-10:
You need to have PhoneGap or Cordova installed. I recommend Cordova. Check out the getting started guide https://cordova.apache.org/#getstarted

closed 2020-04-10 by don

ZebNZ, 2020-04-10:
Hi @don,

**I currently have "Phonegap" installed.**

I am running this command in the Windows CLI:
`phonegap plugin add cordova-plugin-ble-central
`
I get this error:
```
'phonegap' is not recognized as an internal or external command,
operable program or batch file.
```

What am I doing wrong?

Thanks,

Zeb