don/cordova-plugin-ble-central #741: unrecognized selector ios

tjtritonwear, 2020-02-12:
I've added stacktrace info from crashlytics

Fatal Exception: NSInvalidArgumentException
-[CBService value]: unrecognized selector sent to instance 0x2823cb880

CBPeripheral+Extensions.m - Line 195
-[CBPeripheral(com_megster_ble_extension) serviceAndCharacteristicInfo:] + 195

CBPeripheral+Extensions.m - Line 70
-[CBPeripheral(com_megster_ble_extension) asDictionary] + 70

BLECentralPlugin.m - Line 574
-[BLECentralPlugin peripheral:didDiscoverCharacteristicsForService:error:] + 574

CoreBluetooth
(Missing)

UIKitCore
(Missing)

main.m - Line 32
main + 32

libdyld.dylib
(Missing)

don, 2020-04-19:
I'm not sure what to do with this. Did this happen once? Does it happen all the time? Can you provide a simple app that duplicates the problem?

tjtritonwear, 2020-04-19:
Well for starters you're not even checking if didDiscoverCharacteristicsForService is being called from an error, and are just proceeding as if any time that callback is fired it's a good situation.  You're probably going to want to handle the scenario that CoreBluetooth wasn't able to discover the services.

don, 2020-04-19:
> didDiscoverCharacteristicsForService is being called from an error
That's a good catch. I'd appreciate a pull request for the fix.

closed 2024-07-14 by peitschie