don/cordova-plugin-ble-central #745: iOS terminated app due to invalid UUID

vallettea, 2020-03-23:
Hello,

thanks for this great plugin.
I am using it both with android and ios and I have a problem with certain uuids but only on ios.
The error is:

```
Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'String 0000009-0000-4000-0001-000000000000 does not represent a valid UUID'
```

I have log the characteristics from the peripheral object after connection (as suggested [here](https://github.com/don/cordova-plugin-ble-central/issues/270) and it seems good:
```
{properties: ["Read", "Write", "Notify"], isNotifying: false, characteristic: "00000009-0000-4000-0001-000000000000", service: "00000000-0000-4000-0001-000000000000"}
``` 

I have no more idea on how i can debug this. 
Any help appreciated.

vallettea, 2020-03-25:
I'm documenting the error here.
Using Xcode, I was able to have more info:

```
Assertion failure in -[CBUUID initWithString:], /AppleInternal/BuildRoot/Library/Caches/com.apple.xbs/Sources/MobileBluetoothFramework/MobileBluetooth-134.43/CoreBluetooth/CoreBluetooth/CBUUID.m:165
2020-03-25 17:46:26.789492+0100 Gouach[1970:55545] *** Terminating app due to uncaught exception 'NSInternalInconsistencyException', reason: 'String 0000009-0000-4000-0001-000000000000 does not represent a valid UUID'
```

and the error is triggered by line 784 of `BleCentralPlugin.m`:

```
CBUUID *characteristicUUID = [CBUUID UUIDWithString:characteristicUUIDString];
```

According to this [post](https://forums.developer.apple.com/thread/39094) it's quite a old problem and the last thread suggest there is a solution in this [post](https://www.jianshu.com/p/36111b88e1e3) but it's written in chinese and i could only click right and translate but I'm not sure I get it all.

Also during the build there are deprecations warning on CBCentralManager:

[image]

So I think the error is related to this.

vallettea, 2020-03-25:
@don  Basically what I try to know is either this plugin is deprecated on ios or I'm missing something in the way i defined my UUIDS.
thanks for answering. 
Best,

don, 2020-03-25:
The plugin is not deprecated. Your UUID is invalid, it doesn't have enough characters.

vallettea, 2020-03-25:
Thanks for your answer. What do you mean by not enough characters ? Should I replace zeros by a letter ?

halindrome, 2020-03-25:
No.  The UUID is not long enough.  It is not a valid UUID.

vallettea, 2020-03-26:
In fact there is a zero missing in the first part of `0000009-0000-4000-0001-000000000000` 
all my apologies for this but since **android doesn't trigger any error on this** I thought it was something complicated.
thanks for your help.

closed 2020-03-26 by vallettea