don/cordova-plugin-ble-central #746: Running scan when app is in background mode

re2005, 2020-03-25:
This is related to this issue:
https://github.com/don/ble-background/issues/1

Anyone have an solution for that ?

don, 2020-04-19:
iOS should scan in the background if you scan for service UUIDs and set the correct app permissions.

https://developer.apple.com/documentation/corebluetooth/cbcentralmanager/1518986-scanforperipheralswithservices?language=objc

> Your app can scan for Bluetooth devices in the background by specifying the bluetooth-central background mode. To do this, your app must explicitly scan for one or more services by specifying them in the serviceUUIDs parameter. The CBCentralManager scan option has no effect while scanning in the background.

The example app https://github.com/don/ble-background worked in the past and should still work with some updates. It won't work for beacons. Running anything in background on iOS requires more work. I suggest debugging the demo app in Xcode using the latest Cordova, Xcode, and this plugin. If you find an error with the plugin report it back here.

re2005, 2020-04-20:
Hi @don Thanks for the reply.

After long hours trying it, I've realised that.

So if you are advertising BLE from your own device you can indeed set a ServiceUUID, and on IOS it will be able to scan it, when in background mode. It's very slow, and it will sleep after a while, so you may need to restart the scan.

Another thing, if you need to Advertise on background mode, it won't work on iOS, but on Android it works fine.

Cheers

closed 2020-04-20 by re2005