don/cordova-plugin-ble-central #747: Error: Could not find service by UUID

mnavarro-syntonize, 2020-03-27:
Hey everyone,

I'm trying to access to a custom service in my BLE Device but it's giving me an error when trying to write without response:

[image]

The service UUID I'm asking for are OK, but the UUID is not in the advertising, only in Services object, is that the reason why the plugin can't find the custom service? any ideas?

I can access other properties like Firmware or Serial, the problem is only when trying to use a custom service/characteristic.

Thanks in advance!

ASHBAW, 2020-04-09:
I do not know if it is related but I also have a custom service. I connect to our device over bluetooth, start notifications, then send messages down to the device and receive a notification back. I have noticed that this only works about 70% of the time. 

Specifically in a good case I see the following in the Xcode console where data written to the device is prefixed with tx[ and the received data back via the notification is prefixed by rx[:

```
Connected
registering for notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
tx [0x27,0x01,0x01,0x00,0x00,0x0b,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
rx [0x27,0x00,0x01,0x01,0x00,0x00,0x0b,0x39,0x31,0x2e,0x37,0x31,0x2e,0x35,0x36,0x00,0x00,0x00]
tx [0x2c,0x01,0x01,0x03,0x00,0x08,0x07,0xe4,0x04,0x27,0x15,0x08,0x03,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
didUpdateValueForCharacteristic
rx [0x2c,0x00,0x01,0x01,0x03,0x00,0x08,0x07,0xe4,0x04,0x27,0x15,0x08,0x03,0x00]
tx [0x20,0x02,0x01,0x43,0x49,0x0c,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
didUpdateValueForCharacteristic
rx [0x20,0x00,0x02,0x01,0x43,0x49,0x0c,0x43,0x43,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]

```

In the bad case I never receive any data back via the notification and eventually see the **Could not find service with UUID**

```
Connected
registering for notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
tx [0x27,0x01,0x01,0x00,0x00,0x0b,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
ERROR: {"line":1,"column":536453,"sourceURL":"ionic://localhost/main-es2015.bff2283c4723329aede5.js"}
stop notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
stop notification
getData
Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 16
Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 32
stop notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
2020-04-08 16:34:37.542716-0500 CCPCS[677:101095] stop notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
disconnect
Cleared stop notification callback BLE351567158 for key 0CE8DAEE-7059-F1F6-87C0-238297987E89|832AE7FE-D3C5-11E4-B9D6-1681E6B88EC1|C73F5580-91EB-11E5-8994-FEFF819CDC9F
didDisconnectPeripheral
stop notification
getData
Could not find service with UUID 832ae7fe-d3c5-11e4-b9d6-1681e6b88ec1 on peripheral with UUID 0CE8DAEE-7059-F1F6-87C0-238297987E89
disconnect
```

don, 2020-04-19:
@mnavarro-syntonize the service doesn't need to be advertised. If you connected you should find it. You can run into problems if the services and characteristics on your device change because you're also developing the device. The OS can use cached results.

If the characteristics are changing for your service, your peripheral should implement Service Changed 0x2a05 on the Generic Attribute Service 0x1801 so the phone knows not to used cached data.

Also check for typos and case mismatch in UUIDs.

closed 2020-04-19 by don