don/cordova-plugin-ble-central #750: Location Services required to be enabled

jabelcs, 2020-04-09:
First of all many thanks for your great work.

From what I've been reading, this plugin and others alike requires android device Location Services to be enabled in order to find bluetooth sensors.

I have an app in production, made with cordova, and I've been receiving some critics from users saying that the app can't find bluetooth sensors. I realised that the problem was that the Location Services are off on the users devices. Everytime I receive a complaint I have to inform the user that he must enable Location Services, must users don't even know this option exists.

My question is that I have installed another app called "wahoo" that is able to find the bluetooth sensors wether the Location Services are On or Off.

Is this a problem of hybrid apps or a problem of this kind of cordova plugins?

Hope you can give some clues on why this happens and if there's a way to find bluetooth sensors with Location Services disabled.

Thanks in advance for your time.

don, 2020-04-10:
> My question is that I have installed another app called "wahoo" that is able to find the bluetooth sensors wether the Location Services are On or Off.

I think the behavior depends on your phone manufacturer and the Android version. There were problems with is before. I can't recall the details but you should be able to find them searching the issue history.

don, 2020-04-19:
You can use [isLocationEnabled](https://github.com/don/cordova-plugin-ble-central#islocationenabled) to check if location services are enabled. I don't know how you tell if the device needs this enabled to scan.

At one point, the plugin briefly required locations services for scanning, but those changes were reverted. Reading through the the [issues about location services](https://github.com/don/cordova-plugin-ble-central/issues?q=is%3Aissue+location+services) can supply you with more details.

closed 2020-04-19 by don