don/cordova-plugin-ble-central #751: Notifications not always working

ASHBAW, 2020-04-10:
I am seeing this issue more often on iOS devices.  ~30% of the time I fail to receive a notification from the connected device after writing a message to it at which point I timeout and disconnect. Android I rarely see the issue and my connection and communication rate is closer to 100%

After connecting to a device the following way (BleDevice.ConnectionTimeout = 5500)
```
 connect(deviceId: string) {
    return this.ble.connect(deviceId).pipe(
      timeoutOnce({
        timeout: BleDevice.ConnectionTimeout
      }),
      tap(() => {
        this.connectedDeviceId = deviceId;
        this.io.start(deviceId);
      })
    ) as Observable<Peripheral>;
  }
```
I call io.start:

```
  start(deviceId: string) {
    this.deviceId = deviceId;
    this.ble.startNotification(this.deviceId, hickory.id, hickory.characteristics.dataReady)
      .subscribe(rx => this.receive(rx), error => this.rejectTransaction(error));
  }
```

After this is called I start sending messages. 

I captured the issue on Xcode console.

Here is a good case where I send a message down - tx[] - and you can see it is looking for the following asssociated notification - Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8. I then receive the message back via the notification - rx[]

```
Connected
registering for notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
tx [0x27,0x01,0x01,0x00,0x00,0x0b,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
rx [0x27,0x00,0x01,0x01,0x00,0x00,0x0b,0x39,0x31,0x2e,0x37,0x31,0x2e,0x35,0x36,0x00,0x00,0x00]
tx [0x2c,0x01,0x01,0x03,0x00,0x08,0x07,0xe4,0x04,0x27,0x15,0x08,0x03,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
didUpdateValueForCharacteristic
rx [0x2c,0x00,0x01,0x01,0x03,0x00,0x08,0x07,0xe4,0x04,0x27,0x15,0x08,0x03,0x00]
tx [0x20,0x02,0x01,0x43,0x49,0x0c,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
didUpdateValueForCharacteristic
rx [0x20,0x00,0x02,0x01,0x43,0x49,0x0c,0x43,0x43,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
```
Here is a bad case where even though I am looking for the notification like the good case:
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
I eventually timeout after 5.5 seconds - 

`ERROR: {"line":1,"column":536453,"sourceURL":"ionic://localhost/maines2015.bff2283c4723329aede5.js"} `

and disconnect

```
Connected
registering for notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
tx [0x27,0x01,0x01,0x00,0x00,0x0b,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
getData
Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
didUpdateValueForCharacteristic
ERROR: {"line":1,"column":536453,"sourceURL":"ionic://localhost/main-es2015.bff2283c4723329aede5.js"}
stop notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
stop notification
getData
Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 16
Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 32
stop notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
2020-04-08 16:34:37.542716-0500 CCPCS[677:101095] stop notification
getData
Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
disconnect
Cleared stop notification callback BLE351567158 for key 0CE8DAEE-7059-F1F6-87C0-238297987E89|832AE7FE-D3C5-11E4-B9D6-1681E6B88EC1|C73F5580-91EB-11E5-8994-FEFF819CDC9F
didDisconnectPeripheral
stop notification
getData
Could not find service with UUID 832ae7fe-d3c5-11e4-b9d6-1681e6b88ec1 on peripheral with UUID 0CE8DAEE-7059-F1F6-87C0-238297987E89
disconnect
```
Is anyone else seeing this? 

Again Android I really don't see the problem except very rarely. I see it more on iOS devices ~30% of the time.

Is there any work around?

Is how I implemented this incorrect?

Thanks for any help.

ASHBAW, 2020-04-10:
Just a little more information. I added some logging to the code specifically:

```
(void)peripheral:(CBPeripheral *)peripheral didUpdateValueForCharacteristic:(CBCharacteristic *)characteristic error:(NSError *)error {
    NSLog(@"didUpdateValueForCharacteristic");

    NSString *key = [self keyForPeripheral: peripheral andCharacteristic:characteristic];
    NSString *notifyCallbackId = [notificationCallbacks objectForKey:key];
    NSLog(@"didUpdateValueForCharacteristic-BAW1");
    if (notifyCallbackId) {
        NSData *data = characteristic.value; // send RAW data to Javascript
        NSLog(@"didUpdateValueForCharacteristic-BAW2");
        NSLog(@"bytes in hex: %@", [data description]);
        CDVPluginResult *pluginResult = nil;
        if (error) {
            NSLog(@"%@", error);
            pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_ERROR messageAsString:[error localizedDescription]];
        } else {
            NSLog(@"didUpdateValueForCharacteristic-BAW3");
            pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_OK messageAsArrayBuffer:data];
        }
        NSLog(@"didUpdateValueForCharacteristic-BAW4");
        [pluginResult setKeepCallbackAsBool:TRUE]; // keep for notification
        [self.commandDelegate sendPluginResult:pluginResult callbackId:notifyCallbackId];
    }
    NSLog(@"didUpdateValueForCharacteristic-BAW5");
    NSString *readCallbackId = [readCallbacks objectForKey:key];

    if(readCallbackId) {
        NSLog(@"didUpdateValueForCharacteristic-BAW6");
        NSData *data = characteristic.value; // send RAW data to Javascript
        CDVPluginResult *pluginResult = nil;
        
        if (error) {
            NSLog(@"%@", error);
            pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_ERROR messageAsString:[error localizedDescription]];
        } else {
            pluginResult = [CDVPluginResult resultWithStatus:CDVCommandStatus_OK messageAsArrayBuffer:data];
        }
        NSLog(@"didUpdateValueForCharacteristic-BAW7");
        [self.commandDelegate sendPluginResult:pluginResult callbackId:readCallbackId];

        [readCallbacks removeObjectForKey:key];
    }
} 
```
Here is the good case:

```
2020-04-10 09:39:45.391318-0500 CCPCS[503:105095] Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 16
2020-04-10 09:39:45.391907-0500 CCPCS[503:105095] Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 32
2020-04-10 09:39:45.398279-0500 CCPCS[503:105095] registering for notification
2020-04-10 09:39:45.398651-0500 CCPCS[503:105095] getData
2020-04-10 09:39:45.399112-0500 CCPCS[503:105095] Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16
2020-04-10 09:39:45.471384-0500 CCPCS[503:105095] tx [0x27,0x01,0x01,0x00,0x00,0x0b,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00,0x00]
2020-04-10 09:39:45.472190-0500 CCPCS[503:105095] getData
2020-04-10 09:39:45.472375-0500 CCPCS[503:105095] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-10 09:39:45.820763-0500 CCPCS[503:105095] didUpdateValueForCharacteristic
2020-04-10 09:39:45.821061-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW1
2020-04-10 09:39:45.821162-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW2
2020-04-10 09:39:45.821330-0500 CCPCS[503:105095] bytes in hex: {length = 18, bytes = 0x2700010100000b39312e37312e3536000000}
2020-04-10 09:39:45.821447-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW3
2020-04-10 09:39:45.821556-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW4
2020-04-10 09:39:45.821835-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW5
2020-04-10 09:39:45.825448-0500 CCPCS[503:105095] rx [0x27,0x00,0x01,0x01,0x00,0x00,0x0b,0x39,0x31,0x2e,0x37,0x31,0x2e,0x35,0x36,0x00,0x00,0x00]
```
Here is the bad case:

```
2020-04-10 09:33:08.091133-0500 CCPCS[503:105095] Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 16
[... 9 lines cut ...]
2020-04-10 09:33:08.262767-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW5
 
2020-04-10 09:33:09.254584-0500 CCPCS[503:105095] didUpdateValueForCharacteristic
2020-04-10 09:33:09.254850-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW1
2020-04-10 09:33:09.255005-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW2
2020-04-10 09:33:09.255266-0500 CCPCS[503:105095] bytes in hex: {length = 20, bytes = 0x0000000600000000000124000000000000000000}
2020-04-10 09:33:09.255480-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW3
2020-04-10 09:33:09.255666-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW4
2020-04-10 09:33:09.256028-0500 CCPCS[503:105095] didUpdateValueForCharacteristic-BAW5

2020-04-10 09:33:13.668518-0500 CCPCS[503:105095] ERROR: {"line":1,"column":536161,"sourceURL":"ionic://localhost/main-es2015.e6a8352cb4f42f2fe482.js"}
```
In the good case readCallbackId has a valid value the first time through

```
    NSString *readCallbackId = [readCallbacks objectForKey:key];
    if(readCallbackId) { }
```
In the bad case readCallbackId does not have a valid value the first time through and does not go into the if statement. The second time through it does have a value and gets into the if statement but the data I get back at this point is not valid and the callback in my javascript code times out and I disconnect from my device.

Any thoughts on why this is happening?

A work around?

Thanks

don, 2020-04-10:
If you're doing notifications, `didUpdateValueForCharacteristic` should use the `notifyCallbackId ` not the `readCallbackId`.

I recommend writing a very simple test app with just this functionality and only JavaScript. See if you can duplicate the issue. 

I'm guessing that something happens to the notification subscription and it removes the notification callback when the error is called. (You'd need to check the wrapper implementation.)  I see a multiple "stop notification" messages in your log. It think once the subscription fails, it's done. These APIs weren't designed with observables in mind. Often times both the success and failure callback are called multiple times. **If** it turns out that the ionic wrapper doesn't work here you can call the underlying function with callbacks. It's also possible you can re-subscribe if the error happens.

ASHBAW, 2020-04-10:
Thanks for commenting Don.

Yeah it is using notifyCallbackId, that was my mistake.

Yes in any failure case of the subscription(s) it will stop notifications (I have multiple notifications) and disconnect from the device. In my case after a timeout of 5.5 seconds without a notification (I have increased the timeout to 15.5 seconds with it still timing out without a notification)

```
    this.ble.startNotification(this.deviceId, hickory.id, hickory.characteristics.dataReady)
      .subscribe(rx => this.receive(rx), error => this.rejectTransaction(error));
```

I wanted to make sure of your suggestions:

1.  Potentially remove the ionic wrapper which use the Observables (i.e. startNotification(deviceId: string, serviceUUID: string, characteristicUUID: string): Observable<any>;) since the underlying APIs were not designed for this.

2.  Potentially redesign from stopping notifications and disconnecting from the device to re-subscribing / restarting notification in the event of an error/timeout.

3.  Create a simple test case with no wrappers to see if I get the issue where the notifications are not working after starting them.

Just a general question is it normal when working with BLE for notifications to potentially (not common) not work after starting them and you should in your design compensate for this?

Thanks again!

don, 2020-04-19:
The Ionic wrappers automatically calling stopNotification can cause problems. This plugin was designed with callback, many of which can call success and error multiple times. This doesn't work well with Observables since the first error destroys the subscription. (I'm not sure if that is the case with notification)

If you can break this without the ionic wrappers and provide a simple example app, that makes it a lot easier for me to fix.

ASHBAW, 2020-04-20:
Yes I can do that. Thanks.

ASHBAW, 2020-04-24:
Don, I was able to recreate the problem using a simple one page ionic project utilizing the plugin directly without ionic wrappers.

The project can be found here:
https://github.com/ASHBAW/ble_test_one

My project starts two notifications. One is a heartbeat, the other is an indication where a ble.write() is performed (have this set up to perform a ble.write() every second). What I am seeing is that occasionally (more so on iOS) is that I do not ever receive the indication back when I do a ble.write(). I am still receiving the heartbeat notification even in the BAD case.

The console log from Xcode is shown below:
**NOTES:** 

- C73F5580-91EB-11E5-8994-FEFF819CDC9F - heartbeat UUID

- 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 - dataReady indication UUID associated with ble.write()

- 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 - ble.write UUID

**GOOD CASE:**
```
2020-04-23 15:08:18.350641-0500 MyApp[378:25456] Successfully connected

2020-04-23 15:08:18.351208-0500 MyApp[378:25456] registering for notification
2020-04-23 15:08:18.351363-0500 MyApp[378:25456] getData
2020-04-23 15:08:18.351773-0500 MyApp[378:25456] Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16

2020-04-23 15:08:18.352733-0500 MyApp[378:25456] registering for notification
2020-04-23 15:08:18.352875-0500 MyApp[378:25456] getData
2020-04-23 15:08:18.353127-0500 MyApp[378:25456] Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 16
2020-04-23 15:08:18.353505-0500 MyApp[378:25456] Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 32

2020-04-23 15:08:19.348078-0500 MyApp[378:25456] write data
2020-04-23 15:08:19.349152-0500 MyApp[378:25456] getData
2020-04-23 15:08:19.349474-0500 MyApp[378:25456] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-23 15:08:19.419360-0500 MyApp[378:25456] write Callback null

2020-04-23 15:08:19.652080-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 15:08:19.658738-0500 MyApp[378:25456] dataReady Callback 39

2020-04-23 15:08:20.348583-0500 MyApp[378:25456] write data
2020-04-23 15:08:20.349405-0500 MyApp[378:25456] getData
2020-04-23 15:08:20.349746-0500 MyApp[378:25456] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-23 15:08:20.494768-0500 MyApp[378:25456] write Callback null

2020-04-23 15:08:20.760662-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 15:08:20.762018-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 15:08:20.768717-0500 MyApp[378:25456] Hb Callback$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$ 0
2020-04-23 15:08:20.770228-0500 MyApp[378:25456] dataReady Callback 39

2020-04-23 15:08:21.354798-0500 MyApp[378:25456] write data
2020-04-23 15:08:21.355904-0500 MyApp[378:25456] getData
2020-04-23 15:08:21.356232-0500 MyApp[378:25456] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-23 15:08:21.549060-0500 MyApp[378:25456] write Callback null

2020-04-23 15:08:21.660316-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 15:08:21.662715-0500 MyApp[378:25456] dataReady Callback 39

2020-04-23 15:08:22.352174-0500 MyApp[378:25456] write data
2020-04-23 15:08:22.353244-0500 MyApp[378:25456] getData
2020-04-23 15:08:22.353774-0500 MyApp[378:25456] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-23 15:08:22.416590-0500 MyApp[378:25456] write Callback null

2020-04-23 15:08:22.711472-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 15:08:22.720024-0500 MyApp[378:25456] dataReady Callback 39
```

**BAD CASE:**
```
2020-04-23 14:58:55.010039-0500 MyApp[378:25456] Successfully connected

2020-04-23 14:58:55.010613-0500 MyApp[378:25456] registering for notification
2020-04-23 14:58:55.010779-0500 MyApp[378:25456] getData
2020-04-23 14:58:55.011200-0500 MyApp[378:25456] Looking for C73F5580-91EB-11E5-8994-FEFF819CDC9F with properties 16 (HB)

2020-04-23 14:58:55.012261-0500 MyApp[378:25456] registering for notification
2020-04-23 14:58:55.012403-0500 MyApp[378:25456] getData
2020-04-23 14:58:55.012784-0500 MyApp[378:25456] Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 16 (DR)
2020-04-23 14:58:55.013450-0500 MyApp[378:25456] Looking for 93C7A770-DDFF-11E4-B9D6-1681E6B88EC1 with properties 32 (DR)

2020-04-23 14:58:55.472984-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 14:58:55.478438-0500 MyApp[378:25456] Hb Callback$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$ 0

2020-04-23 14:58:56.009423-0500 MyApp[378:25456] write data
2020-04-23 14:58:56.010533-0500 MyApp[378:25456] getData
2020-04-23 14:58:56.011169-0500 MyApp[378:25456] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-23 14:58:56.260862-0500 MyApp[378:25456] write Callback null

2020-04-23 14:58:56.463246-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 14:58:56.467852-0500 MyApp[378:25456] Hb Callback$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$ 0

2020-04-23 14:58:57.010491-0500 MyApp[378:25456] write data
2020-04-23 14:58:57.013266-0500 MyApp[378:25456] getData
2020-04-23 14:58:57.013800-0500 MyApp[378:25456] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-23 14:58:57.095320-0500 MyApp[378:25456] write Callback null

2020-04-23 14:58:57.453360-0500 MyApp[378:25456] didUpdateValueForCharacteristic
2020-04-23 14:58:57.463406-0500 MyApp[378:25456] Hb Callback$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$$ 0

2020-04-23 14:58:58.011527-0500 MyApp[378:25456] write data
2020-04-23 14:58:58.012639-0500 MyApp[378:25456] getData
2020-04-23 14:58:58.013156-0500 MyApp[378:25456] Looking for 737BEA54-D3C5-11E4-B9D6-1681E6B88EC1 with properties 8
2020-04-23 14:58:58.145924-0500 MyApp[378:25456] write Callback null
```

Does this help at all?

I can try going to the next layer of just a Cordova project if this does not help.

ASHBAW, 2020-07-21:
Just following up on this?

ASHBAW, 2020-07-26:
I have spent some more time on this issue. It turns out it was my issue and not the plugins.

On a few of my app pages I continuously write messages. What I found was that if I disconnected in the middle of a write/read (from the notification) the next attempt to connect was not successful. If I stopped the writes and allowed the reads to finish prior to disconnecting the subsequent connection was always successful.

closed 2020-07-26 by ASHBAW