don/cordova-plugin-ble-central #753: How Increase BLE speed

LDLCLid3, 2020-04-21:
Hi guys,

I am working on a project, where I need to do an update in BLE. However, this update is taking a long time, and I would like to lower it.

To give you the characteristics of my project:

- Max data size per packet: 114
-Timeout between each write(): 50ms
- Firmware size: 86 kB
- Time of update now : 2 minutes

Here is an example of the code :

loop method :
`private SendFW(fileBuff: Buffer) {

    return new Promise((sendFWResolve, sendFWReject) => {

      /* STEP 0 : Init transfer with fw bin file size */
      this.fastMsgTransfer(fileBuff.length, DFUSPacket_File_Type.APPLI_FONC).then(
        () => {
          let sendPacket = () => {
            let sendUAR: Uint8Array =
              fileBuff.length - nbdataSent >= DFUSpacket.maxDataSize
                ? new Uint8Array(DFUSpacket.maxDataSize)
                : new Uint8Array(fileBuff.length - nbdataSent);
            fileBuff.copy(sendUAR, 0, nbdataSent, nbdataSent + sendUAR.length);

            /* STEP X : Transfer bin file data */
            this.fastMsgTransfer(sendUAR, DFUSPacket_File_Type.APPLI_FONC).then(
              () => {
                nbdataSent += sendUAR.length;
                this.updateDFUProgress({
                  _text: "dfuUpdate.updatingsaber",
                  _generalProgress:
                    this.curDFUState._generalProgress +
                    packetGeneralProgressStep
                });
                if (nbdataSent < fileBuff.length) {
                  //console.log("Send new packet");
                  sendPacket();
                } else {
                  //console.log("All data sent", nbdataSent);
                  return sendFWResolve();
                }
              },
              reason => {
                console.warn("write DFU Packet Rejected", reason);
                return sendFWReject(reason);
              }
            );
          };
          sendPacket();
        },
        er => {
          console.warn("Init Transfert Error", er);
          this.updateDFUProgress({
            _text: "dfuUpdate.initTransferError",
            _generalProgress: 100
          });
          return sendFWReject(er);
        }
      );
    });
  }`

writeDFUCommand
`protected WriteDFUCommand(sPacket: AbstractSPacket): Promise<void> {
    return new Promise((commandWriteResolve, commandWriteReject) => {

      /* Check if packet is sent in good saber mode */
      if (
        ((sPacket instanceof DFUSpacket && this._isSaberDFUMode) ||
          (sPacket instanceof SPacket &&
            !this
              ._isSaberDFUMode)) /* Reject if sPacket type is not good for current saber mode */ &&
        this.isSaberConnected() == true /* Reject if not connected or already sending command */
      ) {
        /* Reject if sPacket is Invalid */

        this.ble
          .write(
            this._connectedDevice.id,
            Protocol_Params.serviceUUID,
            Protocol_Params.characteristicUUID,
            sPacket.getBytes())
          .then(
            () => {
              //console.log("write ok");
              setTimeout(() => {
                return commandWriteResolve();
              }, 50);
            },
            writeError => {
              console.error("Write rejected", writeError);
              return commandWriteReject();
            }
          );
      }
    });
  }`

Thank you Don for this really cool plugin, and thank you for taking time for my request!

TruyenNV, 2020-05-14:
You can try to request connection priority to 'high' and request Mtu size to 512 on Android via ble.requestConnectionPriority and ble.requestMtu methods.
Could you please send your upgrade FW code for my reference? I'm also doing it.

don, 2020-11-08:
BLE is really slow especially for firmware updates. Bluetooth 5 should help. Closing this issue since it's old and I hope you solved it.

closed 2020-11-08 by don