don/cordova-plugin-ble-central #755: Plugin doesn't scan devices when phone's screen is off

n1705771, 2020-04-29:
Hi Sir,

I applied this plugin in my ionic 3 application. I enable the application working in background mode. The plungin's scan method works well when application is in both foreground and background. But when phone's screen turns off, it will stop scanning peripheral devices. Could you please tell how to let plugin works even when phone's screen turns off?

Many thanks

FLATT-Dev, 2020-05-23:
This is not a plugin issue but rather Android issue - the way it manages background application.
You'll need to install  cordova-plugin-background-mode or similar to make sure that your app runs in the background

don, 2020-11-08:
For Android, v1.3.0 of the plugin prompts the user for ANDROID_FINE_LOCATION and ANDROID_BACKGROUND_LOCATION so you might have better luck. For iOS you need to define UIBackgroundModes. This project https://github.com/don/ble-background worked years ago, but iOS may require additional permissions today.  Look into the cordova-plugin-background-mode plugin @FLATT-Dev mentioned.

closed 2020-11-08 by don