don/cordova-plugin-ble-central #756: Reading encrypted characteristic fails on iOS even after pairing

hschultz1960, 2020-05-05:
The following sequence is the general approach to perform implicit pairing at the BLE level.

With the peripheral in pairing mode:

- ble.scan() for peripheral.
- ble.connect() to peripheral using MAC (Android) or UUID (iOS).
- ble.read() characteristic requiring encrypted transport - this forces the OS to display a pairing dialog to the user.
- ble.disconnect().

Afterwards, _without_ the peripheral being in pairing mode, the exact same sequence supposedly can be used, without the OS requiring the user to pair. And in fact, this works fine all the way on Android.

On iOS, the pairing dialog is shown and the initial pairing succeeds. However, subsequent ble.connect() attempts hang. The funny thing is, the peripheral _does_ show up in the list of paired devices under iOS Settings.

Anybody seen this before? 
Is there anything in the above sequence that should be done differently?
Any other call that must be employed for iOS to pick up a previously paired device?

Thanks for any insight you can provide.

hschultz1960, 2020-11-09:
I have an update to this issue, thanks to the help of the manufacturer of the BLE device I am trying to use. It will likely help others to diagnose similar problems. They wrote:

_In BLE v 4.2 there are two major classes of device addresses, namely the Public and the Random device addresses. The public addresses differ from the random ones in that they consist half of a company identifier while the Random ones are - with the exception of two bits - completely random. On the module we use the Random Static address which is a subclass of the random device addresses. It seems that iOS does not persistently store keys when using random static addresses on the module side. This assumption is supported by the observation that we do not detect any attempt on the BLE device's debug output of the iOS device to initiate a connection with the BLE device. Consequently we assume iOS does not support bonding in case of Random static addresses.  We were not aware that this kind of issue may exist and where the error exactly originates from. However, the bug could be quick-fixed by using Public addresses on the module side as we were able to prove iOS functionality when operating with public addresses._

In other words, the device manufacturer acknowledged the problem, and we have since tested that the problem disappears (bonding on iOS works as expected) when using Public MAC addresses. I am aware that not everyone will be able to pursuade the device manufacturer to change their MAC address scheme, but at least we now know what caused the problem.

More information on MAC address classes can be found here https://www.novelbits.io/bluetooth-address-privacy-ble/

don, 2020-11-09:
Thanks for the update. If the device is paired to iOS you _might_  be able to get it using [connectedPeripheralsWithServices](https://github.com/don/cordova-plugin-ble-central#connectedperipheralswithservices) or 
[peripheralsWithIdentifiers](https://github.com/don/cordova-plugin-ble-central#peripheralswithidentifiers).

hschultz1960, 2020-11-09:
I did in fact try those two functions, but to no avail. 

The problem was also verified using the Nordic **nRF Connect** app, which uses native iOS APIs, so the problem is definitely outside of the Cordova plugin.

closed 2022-01-24 by peitschie