don/cordova-plugin-ble-central #757: Slow connection times

FLATT-Dev, 2020-05-06:
My mobile app uses ble-central plugin and a custom device -  I have full control of the firmware. The device  sends 1 BLE packet on 3 channels with 2 second interval. 
Occasionally the app needs to connect to a device and write a characteristic.
I see a widely varying connection times across mobile devices - from 3 seconds to 20-30 seconds - i.e from the time the connect() API is called until the time the connect callback is received. It averages out to about 8-10 seconds.
One other thing. The app continues to scan during the connection attempt. Will stopping the scan while the connection is in progress, help ?
Any suggestions how to improve connection time are greatly appreciated
Thanks

peitschie, 2021-11-17:
Hi @FLATT-Dev 

That connection time variance is expected given the decidedly long interval between advertising packets. Most devices tend to advertise at intervals between 15ms => 200ms between packets.

Stopping the scanning while connecting won't make any impact to this. If you need a faster connection, the only way to achieve that from a mobile phone is to advertise more frequently.

If you are running on Android, there is a possibility that using the Aggressive Scan option (feature coming in a future PR, https://github.com/don/cordova-plugin-ble-central/pull/835) might help find the device more rapidly... but I' don't know that for certain.

I'll keep this open until https://github.com/don/cordova-plugin-ble-central/pull/835 lands, as that's the only plugin change I can think of which might change the connection time here.

peitschie, 2022-02-07:
Options have now been released in 1.4.1 to allow more aggressive scanning.
As noted, further improvements really need much faster advertising intervals on the peripheral, and can't be helped on the phone side, unfortunately.

closed 2022-02-07 by peitschie