don/cordova-plugin-ble-central #760: How to get txPower?

hamzadiaz, 2020-05-08:
I managed to display scanned devices and I can see the RSSI of each device, but I can't find the txPower. This parameter has two definitions:

1. The transmitted power of the peripheral.
2. The received power 1 meter away from the peripheral. (RSSI at 1 meter)

Any Ideas how to calculate it?

Thanks,

hamzadiaz, 2020-05-11:
Update,

I figure out how to translate Byte to string using the `parseAdvertisingData(buffer)` that you provide and this the advertising information that I get:
[image]

peitschie, 2022-01-24:
Just noting, it's very likely likely those are not actually empty, and are instead being displayed as `{}` due to `parseAdvertisingData` returning typed arrays or array buffers, but the formatter is trying to render this an object with no keys.

I'm closing this out as there's nothing obviously different the plugin can do here. But, feel free to re-open if you are still needing help with this!

closed 2022-01-24 by peitschie