don/cordova-plugin-ble-central #762: Can't do writeWithoutResponse on iOS

a-sac, 2020-05-13:
Hello,

So, here is the code:

```
this.ble.writeWithoutResponse(address, "1234", "ABCD", arrayBuffer).then(
      x => this.setStatus("answer to write " + x),
      err => this.setStatus("erro here " + err)
);
```

This works with no problems on Android and it returns OK. When I execute the same code on iOS it returns null and I don't receive anything on the peripheral side.

a-sac, 2020-05-14:
[Update]

So this works between Androids. Despite returning null, it also works between iPhones.
When Android calls `writeWithoutResponse` and the peripheral side is an iOS, it works as well. It still fails when iOS calls `writeWithoutResponse` and the peripheral side is an Android.

don, 2020-05-14:
Check the properties on the characteristic. Android will let you` write without response` to a characteristic that is `write`. iOS will make you use `write`.

a-sac, 2020-05-14:
Yes, you were right. Thank you! :D 
I'm using another ble library to build the peripheral side and my code was:

```
this.bluetoothle.addService({
      service: "1234",
      characteristics: [
        {
          uuid: "ABCD",
          permissions: {
            read: true,
            write: true,
            //readEncryptionRequired: true,
            //writeEncryptionRequired: true,
          },
          properties : {
            read: true,
            writeWithoutResponse: true,
            write: true,
            notify: true,
            indicate: true,
            //authenticatedSignedWrites: true,
            //notifyEncryptionRequired: true,
            //indicateEncryptionRequired: true,
          }
        }
      ]
    })
```

It seems there is another option named `writeNoResponse` that was missing on properties. It's working now!

closed 2020-05-14 by a-sac