don/cordova-plugin-ble-central #763: Read bluetoorh serial success callback function problem

reganGao, 2020-05-14:
self.cordova.ble.read('19:11:28:21:01:08', "FFE0", "FFE1",
        function (buffer) {
          const len = buffer.byteLength; // 128
          const data = new Uint8Array(buffer)  // 0,0,0,0,0,0,0,0,  ... length 128
          const jsonStr = JSON.stringify(buffer) // {}
        },
        function () {
          alert("recv failed");
        }
      );

i promise the bluetooth device had send buffer to my android phone . but i can not get recv 
who can help me ?? 
please!!!

reganGao, 2020-05-14:
i can write to this address service device success; but can not get recv..

don, 2020-05-14:
Stackoverflow is a better place for a question on how to use the library. You need to make sure the FFE1 allows reading. Typically with something like this you'd want to subscribe to notifications on FFE1 rather than reading.

closed 2020-05-14 by don