don/cordova-plugin-ble-central #765: Manufacturer specific Data

buildog, 2020-05-20:
According to the lib (https://github.com/don/cordova-plugin-ble-central#ios-1) here is the way to get this data
```
const mfgData = new Uint8Array(device.advertising.kCBAdvDataManufacturerData);
console.log('Manufacturer Data: ', mfgData);
```

Then parse the bytes to hex string via 
```
const hex = Buffer.from(mfgData).toString('hex');
console.log(hex);
``` 

The encoded to hex string result being `2604 0504 386 55c0b`

What I don't understand is the manufacturer specific data is supposed to start with the company specific ID and in my case this id is `0x0426` (hex) or `1062` (decimal) while my string start with `2604`

Am I doing something wrong with my parsing ?

hschultz1960, 2020-05-20:
Looks like you are being hit by the classic byte ordering issue ... bytes are being transmitted over the air in so-called "network byte order", whereas on an x86 architecture system, bytes are stored in so-called "host byte order". (Google "endianess".) So your 0x2604 consists of 0x26 and 0x04, which swapped becomes 0x0426, which is exactly what you are looking for.

buildog, 2020-05-20:
Thanks @hschultz1960 

On the other side I think this adv is also used to read a serial number, which is `190604600` but I can find any sequence, either partially,  which fit this serial in the parsed string `2604 0504 386 55c0b` 

Here is a packets dump 

``` 
May 18 15:25:45.280  HCI Command      0x0000                     LE Set Scan Enable - 0x00, Filter Duplicates: 1  0C 20 02 00 01   
May 18 15:25:45.280  Power            0x0000                     H4 wake  48 34 20 77 61 6B 65   
May 18 15:25:45.287  HCI Event        0x0000                     Command Complete [200C] - LE Set Scan Enable  0E 04 01 0C 20 00   
May 18 15:25:45.287  HCI Command      0x0000                     Vendor Specific Command  E9 FC 09 05 02 00 00 00 00 00 00 02   
May 18 15:25:45.287  HCI Event        0x0000                     Command Complete [FCE9] -   0E 06 01 E9 FC 00 05 02   
May 18 15:25:45.287  HCI Command      0x0000                     Vendor Specific Command  E9 FC 09 07 02 00 00 00 00 00 00 02   
May 18 15:25:45.287  HCI Event        0x0000                     Command Complete [FCE9] -   0E 07 01 E9 FC 00 07 02 08   
May 18 15:25:45.287  HCI Command      0x0000                     Vendor Specific Command  E9 FC 01 1E   
May 18 15:25:45.288  HCI Event        0x0000                     Command Complete [FCE9] -   0E 07 01 E9 FC 00 1E 1E 1E   
May 18 15:25:45.288  HCI Command      0x0000                     Vendor Specific Command  E9 FC 02 35 02   
May 18 15:25:45.288  HCI Event        0x0000                     Command Complete [FCE9] -   0E 08 01 E9 FC 00 35 02 1E 1E   
May 18 15:25:45.288  HCI Command      0x0000                     LE Set Scan Parameters - Active - 30/40 (ms)  0B 20 07 01 40 00 30 00 01 00   
May 18 15:25:45.288  HCI Event        0x0000                     Command Complete [200B] - LE Set Scan Parameters  0E 04 01 0B 20 00   
May 18 15:25:45.288  HCI Command      0x0000                     LE Set Scan Enable - 0x01, Filter Duplicates: 1  0C 20 02 01 01   
May 18 15:25:45.288  HCI Event        0x0000                     Command Complete [200C] - LE Set Scan Enable  0E 04 01 0C 20 00   
May 18 15:25:45.297  HCI Event        0x0000  C4:64:E3:9B:04:44  LE - Advertising Report - 1 Report - Normal - Public - C4:64:E3:9B:04:44  -81 dBm - Channel 37  3E 21 02 01 00 00 44 04 9B E3 64 C4 15 02 01 06 11 06 E4 6D DC 75 BF DD E5 84 1A 42 0E 0B 01 00 BD 98 AF   
May 18 15:25:45.297  HCI Event        0x0000  C4:64:E3:9B:04:44  LE - Advertising Report - 1 Report - Normal - Public - C4:64:E3:9B:04:44  -83 dBm - Manufacturer Specific Data - 315X - Channel 37  3E 1C 02 01 04 00 44 04 9B E3 64 C4 10 09 FF 26 04 05 04 38 65 5C 0B 05 08 33 31 35 58 AD   
May 18 15:25:45.490  Power            0x0000                     H4 sleep  48 34 20 73 6C 65 65 70   
May 18 15:25:48.282  HCI Command      0x0000                     LE Set Scan Enable - 0x00, Filter Duplicates: 1  0C 20 02 00 01   
May 18 15:25:48.283  Power            0x0000                     H4 wake  48 34 20 77 61 6B 65   
May 18 15:25:48.285  HCI Event        0x0000                     Command Complete [200C] - LE Set Scan Enable  0E 04 01 0C 20 00   
May 18 15:25:48.286  HCI Command      0x0000                     LE Set Scan Parameters - Active - 30/40 (ms)  0B 20 07 01 40 00 30 00 01 00   
May 18 15:25:48.287  HCI Event        0x0000                     Command Complete [200B] - LE Set Scan Parameters  0E 04 01 0B 20 00   
May 18 15:25:48.287  HCI Command      0x0000                     LE Set Scan Enable - 0x01, Filter Duplicates: 1  0C 20 02 01 01   
May 18 15:25:48.287  HCI Event        0x0000                     Command Complete [200C] - LE Set Scan Enable  0E 04 01 0C 20 00   
May 18 15:25:48.364  HCI Event        0x0000  C4:64:E3:9B:04:44  LE - Advertising Report - 1 Report - Normal - Public - C4:64:E3:9B:04:44  -79 dBm - Channel 39/38  3E 21 02 01 20 00 44 04 9B E3 64 C4 15 02 01 06 11 06 E4 6D DC 75 BF DD E5 84 1A 42 0E 0B 01 00 BD 98 B1   
May 18 15:25:48.364  HCI Event        0x0000  C4:64:E3:9B:04:44  LE - Advertising Report - 1 Report - Normal - Public - C4:64:E3:9B:04:44  -80 dBm - Manufacturer Specific Data - 315X - Channel 39/38  3E 1C 02 01 24 00 44 04 9B E3 64 C4 10 09 FF 26 04 05 04 38 65 5C 0B 05 08 33 31 35 58 B0   
May 18 15:25:48.488  Power            0x0000                     H4 sleep  48 34 20 73 6C 65 65 70 
```

hschultz1960, 2020-05-20:
Same principle: Guessing that your serial number is a 32-bit integer number, it takes 4 bytes. The last 4 bytes of your packet is 0x38655c0b. Reverse those 4 bytes to 0x0b5c6538, and convert to decimal: 190604600. Voila!
In front of your serial number are two bytes: The first is a length byte 0x05, specifying that the following field is 5 bytes long. The next byte is 0x04, which I am guessing signifies that the field value is the serial number. And then follows the 4 bytes as explained above.

buildog, 2020-05-20:
Wow, you saved my day @hschultz1960 ! 🤯
Thanks a lot !

closed 2020-05-20 by buildog