don/cordova-plugin-ble-central #766: device disconnects immediately after connecting

ClaeysJoke, 2020-05-20:
```
Hello! I am trying to connect with a temperature measuring device from Andesfit. However, after connecting with the device, it's immediately disconnected. Can any of you help? Thank you!

var app = {
    initialize: function() {
        document.addEventListener('deviceready', this.onDeviceReady.bind(this), false);
    },
    onDeviceReady: function() {
        this.receivedEvent('deviceready');
        this.startScanForDevices();
        //ble.autoConnect("B4:52:A9:C5:9D:6E",app.onConnected,app.onDisconnected);
    },
    receivedEvent: function(id) {
        var parentElement = document.getElementById(id);
        var listeningElement = parentElement.querySelector('.listening');
        var receivedElement = parentElement.querySelector('.received');

        listeningElement.setAttribute('style', 'display:none;');
        receivedElement.setAttribute('style', 'display:block;');

        console.log('Received Event: ' + id);
    },

    onDiscoverDevice: function(device){
        var devicediscovered=0;
        console.log(JSON.stringify(device));
        var listItem = document.createElement('li');
        listItem.innerHTML = device.name + '<br/>' +
            device.id + '<br/>' +
            'RSSI: ' + device.rssi;
        deviceList.appendChild(listItem);

        if(device.name=='TEMP'){
            console.log('temp found');
            console.log(device.id);
            ble.connect(device.id,app.onConnected,app.onDisconnected);
        }
    },

    onDisconnected:function(device){
        console.log('disconnected temp device');
    },

    onDisconnectedState:function(device){
        console.log('disconnected temp device state');
    },

    onConnected: function(device){
        console.log('connected with temp device');
        ble.isConnected(device.id, app.onConnected, app.onDisconnectedState);
        console.log(JSON.stringify(device));
        console.log('trying to start notification')
         ble.startNotification(device.id, "1809", "2A1C", app.onNotificationStarted, function() {
             console.log('start notification failed');
        });
    },

    onNotificationStarted: function(buffer){
        console.log('notification started');
        var data = new Uint8Array(buffer);
        console.log("Temp state changed to " + data[0]);
    },

    onNotification2Started: function(buffer){
        var data = new Uint8Array(buffer);
        console.log("Temp state 2 changed to " + data[0]);
    },

    startScanForDevices: function() { 
        console.log('scanning started');
        ble.startScan([], app.onDiscoverDevice, function() {
            console.log('scanning failed');
        });
        setTimeout(ble.stopScan,
            5000,
            function() { console.log("Scan complete"); },
            function() { console.log("stopScan failed"); }
        );
     } 
};

app.initialize();
```

ClaeysJoke, 2020-05-20:
Update: if I remove "ble.startnotification", I get the feedback that it is disconnected from "onDisconnectedState" but not from "onDisconnected"

don, 2020-06-22:
The device probably expects an initialization message after you connect. If there's no documentation, try using the manufacturers app and sniffing the bluetooth communication to see what it's doing.

benplain, 2021-01-24:
@ClaeysJoke,

I found that with my devices (heart rate monitors), when I used `ble.connect` it dropped the connection instantly. However, when I use `ble.autoConnect`, it keeps the connection open and I don't get the instant disconnect.

I'm not sure if this is the correct way to do it but it works for me.

Gargamil, 2021-11-24:
> @ClaeysJoke,
>
> I found that with my devices (heart rate monitors), when I used `ble.connect` it dropped the connection instantly. However, when I use `ble.autoConnect`, it keeps the connection open [...]

I have exactly the same problem, but only on Android. iOS works with "normal" connect perfectly.